espressif/esp-at #75: AT Commands Firmware with AWS TLS/SSL Client connection

schwac, 2018-03-07:
Hi, I have a post going over at ESP32.net as well, but I wanted to check in here since this forum is focused on ESP32 AT command firmware. 
https://www.esp32.com/viewtopic.php?f=2&t=4901

Basically I am trying to connect to AWS IoT with the latest ESP32 AT cmd firmware, and I cannot get it to successfully complete the SSL hand shake. I am nearly positive that I have loaded the certs correctly, and tested them with the SSL windows command line utility, but it will not work on the ESP32. There are many more details on that other thread. 

I would really appreciate some help on this and thought that there may be some AT Cmd experts watching this repo and could confirm that they have indeed connected an ESP32 running the AT firmware to AWS at some point? Thanks in advance!

xcguang, 2018-03-12:
Hi @schwac ,
Please refer to https://github.com/espressif/esp32-at/blob/master/tools/README.md about ```2. PKI Bin```

schwac, 2018-03-12:
Ah ha! Thanks for referring me to that section. I just used that tool to generate the at_cert.bin, at_key.bin, and at_ca.bin and tried loading them to both the server and client locations - neither of these worked. Same error as before... I will keep trying various combinations. Have you got the AT cmd FW to connect to AWS IoT before?

schwac, 2018-03-12:
I can confirm that this does not work if I load them to either the server locations (CERT = 0x24000, KEY 0x26000 and CA =0x28000) or the client locations (0x2A000, 0x2C000, 0x2E000). I have inspected the actual binaries created by the AtPKI python tool and they seem to be correct.

xcguang, 2018-03-13:
Hi,
We haven't connected to AWS IoT, but we have tested ssl server and client with cert and ca.
Could you give some log from UART0?

schwac, 2018-03-18:
Have you had a chance to check out that UART log file yet?

xcguang, 2018-03-20:
Hi,
```c
I (23906) openssl: ssl_tls.c:2441 ssl->f_send() returned -80 (-0x0050)

I (23912) openssl: ssl_tls.c:2846 mbedtls_ssl_flush_output() returned -80 (-0x0050)

I (23920) openssl: ssl_tls.c:4710 mbedtls_ssl_write_record() returned -80 (-0x0050)

I (23929) openssl: mbedtls_ssl_handshake() return -0x50
ssl connect fail
```
Maybe socket is reset by server in ```mbedtls_net_send```. The error no is defined as 
```#define MBEDTLS_ERR_NET_CONN_RESET                        -0x0050  /**< Connection was reset by peer. */```

schwac, 2018-03-20:
thanks for taking a look... do you know if I should be loading the key, cert and ca bin files into the client positions or the server positions?

xcguang, 2018-03-20:
If you create as ssl client by "AT+CIPSTART" command , the key, cert and ca bin files should download into 0x2c000, 0x2a000 and 0x2e000.
The following is the partition information
```
# Name	       Type SubType	Offset	Size
client_cert	0x40	5	0x2a000	8K
client_key	0x40	6	0x2c000	8K
client_ca	0x40	7	0x2e000	8K
```
Please pay attention to the size of the file. And you can adjust offset and size if necessary.

schwac, 2018-03-20:
I still cannot get this to work... I have loaded the binaries into the proper locations and it just won't connect to AWS via wifi. It programs fine and looks like it it starting the TLS handshake. Same sort of failure code as aIl the other  work on them. 
[image]
[esp32_at_SSL_AWS_Connect_Fail_03-20-18.txt](https://github.com/espressif/esp32-at/files/1828571/esp32_at_SSL_AWS_Connect_Fail_03-20-18.txt)

Any recommendations would be appreciated!.

xcguang, 2018-03-20:
@schwac ,
There is a lack of a command in the document, and sorry for that.
#### AT+CIPSSLCCONF — Config SSL client
```
Single connection: (+CIPMUX=0)
AT+CIPSSLCCONF=<auth_mode>,<pki_number>,<ca_number>
Multiple connections: (+CIPMUX=1)
AT+CIPSENDEX=<link ID>,<auth_mode>,<pki_number>,<ca_number>
```
Parameters:  
- \<link ID>: ID of the connection (0~max), for multiple connections, if the value is max, it means all connections. By default, max is 5.  
- \<auth_mode>:   
    - 0: no authorization.
    - 1: load cert and private key for server authorization.
    - 2: load CA for client authorize server cert and private key.
    - 3: both authorization.  

- \<pki_number>: the index of cert and private key, if only one cert and private key, the value should be 0.  
- \<ca_number>: the index of CA, if only one CA, the value should be 0.
## Notes:
- Call this command before establish SSL connection if you want configuration take effect immediately.
- The configuration changes will be saved in the NVS area. If you use AT+SAVETRANSLINK to set SSL passthrough mode, ESP32 SSL will be connected based on this configuration after power on.

schwac, 2018-03-20:
Oh man! Yes, it sounds like that command is critical. I will test it ASAP. 

Also, I think there is a typo in your example above: The command is AT+CIPSSLCCONF for both the single and multiple connections right? Not AT+CIPSENDEX?

Also, when will the next AT release be that contains SSL support for clients, etc?

xcguang, 2018-03-21:
Yes, The command is  AT+CIPSSLCCONF. 
Now we are testing esp32-at, we will release it after testing Ok.

schwac, 2018-03-21:
Awesome! That worked! I am connected successfully. 

One more question: Can I make two or more client SSL connections simultaneously?

xcguang, 2018-03-21:
Yes, you can if ram is enough, otherwise, it will fail.

schwac, 2018-03-21:
How do I know how much RAM is used? Can I conserve RAM by excluding certain features during build?

xcguang, 2018-03-21:
You can send ```AT+SYSRAM?``` to query remained RAM.

schwac, 2018-03-21:
Cool. I consider this closed! Thank you so much for helping out!

closed 2018-03-21 by schwac

tcpipchip, 2019-08-06:
works fine on NINA W102

Wtpaul, 2019-09-25:
Hi schwac
I am also trying to connect to aws iot via wroom32 with AT commands.
Do you have any docs / advice on how to build the packets?
i have connected to a local server with ssl... but i was in charge of data structure at both ends.  it is hard to find aws docs that don't use the freertos code or aws sdk command line.  Any help would be appreciated.

tcpipchip, 2019-09-25:
Did you create the 3 certificates and burn them ?
[image]
0x27000 client-cert.bin
0x28000 private_key.bin
0x29000  amazon-ca.bin
[image]
Port 8443
AT+CIPSTART="SSL","xxxxxxxxxxxxxx-ats.iot.us-east-1.amazonaws.com",8443
cert_len=1224
prvtkey_len=1679
ca_len=1188
ctx ok
CONNECT
OK
[image]
And then
POST /things/lamp/shadow HTTP/1.1
Host: xxxxxxxxxxxxxx-ats.iot.us-east-1.amazonaws.com
Content-Type: application/json  
Content-Length: 89

{
    "state": {
        "reported": { "temp":"12","umidade":"88"
        }
    }
}

Wtpaul, 2019-09-25:
Not yet..
when you create a new certificate on AWS you can download 
certificate.pem, public.pem, private.pem and a root CA for AWS IOT
so...
i keep the private.pem in a safe place
how do i convert the other 3 into *.bin files?

tcpipchip, 2019-09-25:
See item 2
https://esp32.com/viewtopic.php?t=9374

tcpipchip, 2019-09-25:
it´s a python utility! Works very nice!

Wtpaul, 2019-09-27:
Hi tcpipchip
Thank you for the detail above... I have successfully read data from the AWS iot shadow of my thing.

One small issue.. maybe you can help with.
I'm running the following version of AT command firmware on the esp-wroom-32
> SDK version:v3.0.3                      
> compile time:Sep  4 2018 08:53:40                          
> Bin version(Wroom32):1.1.2 

When i try to read my aws shadow thing i get the following message from the usb debug port.
> cert_len=1220
> prvtkey_len=1679
> ca_len=1206
> ctx ok
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:
> CPU 0: at_process_task

basically i looks like code on the ESP is stuck in a loop too long and causes a watchdog issue.
can this be fixed ... or is it a non issue.

issue 2
Also if i do a firmware upgrade 
it upgrades to 
> SDK version:v3.2-127-gddbce78                              
> compile time:Jun 27 2019 11:21:27                          
> Bin version:1.2.0(WROOM-32) 

but this message appears in the debug output

> factory_parameter partition missed
> 1.2.0
> factory_parameter partition missed

is this a problem?

Many thanks for all your help
William

tcpipchip, 2019-09-27:
well, 
About the wdt, see if running 240Mhz !
About the second...
I found something 
https://github.com/espressif/esp-at/blob/master/main/at_default_config.c
Are you using WROOM ?

Wtpaul, 2019-10-04:
Hi tcpipchip
Lots of progress this week.
Got esp working at 240MHz.. fixed the issue above.. fresh compile of latest git code.
I can connect via REST API to the AWS shadow and do a read & write...  
Thanks for all the help.

I'm now trying to get the MQTT working....
so i have the options enabled for the AT command

> AT_MQTT_USERCFG=0,3,"wtp","","",0,0,""
ie Link 0, MQTT over TLS(verify server cert), client ID= wtp, no username, no password, cert_key=0, CA_ID=0, no path.
i don't think you need username and password for AWS (is this correct??)

> AT_MQTT_CONNCFG=0,30,0,"lwt_topic","lwt_msg",0,0
ie link 0, keepalive for 30s, clean session, lwt....etc

This is the debug output
> [at-mqtt]warning code:24600
> [at-mqtt]warning code:24603
> all parameters are ready
> mqtt_ca format error:ffff
the errors are letting me know the username and password are null.
I think there is a separate cert_key & CA_ID from the client cert,etc for https???
where do i find info on how to generate this cert, if it is different from before.. and where do i find the address table to know where to upload it to?

Many Thanks for your patience with my questions...

tcpipchip, 2019-10-05:
No necessary user pass to mqtt!
We have to know the difference between server-certifcate and client-certificates.
The address are 
[image]

tcpipchip, 2019-10-06:
Look this curl server certificate

tcpipchip, 2019-10-06:
https://www.thingforward.io/techblog/2018-07-18-https-on-the-esp32-server-and-client-side.html

Wtpaul, 2019-10-07:
Sorry... is username and password necessary when connecting to AWS via MQTT?
if so, where are these found?

In the at_customise.csv (see attached) there are memory locations defined for mqtt_certificates, keys and  CA
[image]

Are these the same certificates as the client certificates, keys and CA?
Thank you

tcpipchip, 2019-10-07:
I never used user/pass to aws iot mqtt

I remember that are the same certificates for both, https and mqtt

[at_customize_csv]<https://user-images.githubusercontent.com/39193764/66295650-b8f82080-e8e3-11e9-8eaf-3229f192e783.JPG>

tcpipchip, 2019-10-07:
I never used user/pass to aws iot mqtt

I remember that are the same certificates for both, https and mqtt.

But i never tested mqtt protocol on ESP32 AT COMMANDS

xcguang, 2019-10-07:
https://blog.csdn.net/espressif/article/details/101714951 is a demo to connect AWS using `AT+MQTT`. But it is written in Chinese.

tcpipchip, 2019-10-07:
Hi @xcguang 

Interesting...but...where did they get user and pass ?

tcpipchip, 2019-10-07:
looks let you have to change the file contents and compile again, right ?
AmazonRootCA1.pem 替换 components/customized_partitions/raw_data/mqtt_ca/mqtt_ca.crt
xxx.cert.pem 替换 components/customized_partitions/raw_data/mqtt_cert/mqtt_client.crt
xxx.private.key 替换 components/customized_partitions/raw_data/mqtt_key/mqtt_client.key

xcguang, 2019-10-07:
Compiling is just to get cert, ca and key bin file for AT. 
However, now you cannot get official AT bin, which is supporting `AT+MQTT`.

@ustccw Please comment something about user and password.

tcpipchip, 2019-10-07:
@Wtpaul 
I guess that you will need all informations soon!

ustccw, 2019-10-07:
@xcguang @tcpipchip `username` and `password` is optional for AWS connection, the most important is the `device certificate` and `endpoint`.

tcpipchip, 2019-10-07:
Nice!!!!! Thanks team!

@Wtpaul 
Now you can solve the mqtt problem! Btw, did you compile to 4mb flash our 2mb flash ESP32 ?

Wtpaul, 2019-10-07:
Thanks to all... I have a successful connection with MQTT!! yea!!
I have not taken time to subscribe or publish yet.. will get to that very soon.
Thank you @xcguang for the Chinese link and to google translate :) .
@tcpipchip my ESP-wroom32 is a 4MB device.

tcpipchip, 2019-10-07:
Can you put instruction how to download and compile esp32-at to wroom32 and steps to make it work with mqtt ?

xcguang, 2019-10-08:
`make menuconfig` > `Component config` > `AT` > `[*] AT MQTT command support.`,
then `make`

Wtpaul, 2019-10-08:
The attached pdf has the AT commands i used to connect to AWS IOT with MQTT

[ESP-wroom-32_at_commands.pdf](https://github.com/espressif/esp-at/files/3701559/ESP-wroom-32_at_commands.pdf)

tcpipchip, 2019-10-08:
Is there some tip to compile ?
I followed 
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md (WROOM partition config)
But, i am always getting

tcpipchip, 2019-10-08:
no AT prompt
[image]

Wtpaul, 2019-10-08:
the at commands are via the UART (pins 16&17 )
not via the USB

tcpipchip, 2019-10-08:
hummm, nice! Where is in the make config is this parameter ?

xcguang, 2019-10-09:
@tcpipchip Maybe https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md can help you.

tcpipchip, 2019-10-09:
Hi @xcguang 

Thank you so much! Then my app is really running, only not outputing to the right way

tcpipchip, 2019-10-09:
I cand see the prompt!
but no ECHO when i try to type something. No answer.
Some handshaking that is blocking  ? (RTS, CTS)....

tcpipchip, 2019-10-09:
[image]

xcguang, 2019-10-09:
Maybe you can try https://github.com/espressif/esp-at/issues/117

tcpipchip, 2019-10-09:
no solved, strange that there is not ECHO  when i type a AT command.

Strange! I am using the same serial of LOG

tcpipchip, 2019-10-09:
do you use handsacking ?

tcpipchip, 2019-10-09:
still tryng to make work the rcv to at command.
About
`osboxes@osboxes:~/esp-at/module_config$ ls
module_esp32_default  module_esp8266_default  module_wrover-32
osboxes@osboxes:~/esp-at/module_config$`

where i download module_esp32_wroom ?

tcpipchip, 2019-10-10:
@Wtpaul 
I will need you help! See that my problem! I compiled to a WROOM (esp32_default) and to NINA W102 (2Mb), in both i see the Ready message, but i cant type anything, no echo! Can you report step to step how to config the TXD/RXD log and TXD/RXD at console! 

Thank you!

Wtpaul, 2019-10-10:
I have not encountered this as i have connected to the uart (pins 16&17) on esp32_DevKitc_v4.

Two things to try.
- disable the log output - maybe the secondary stream(ie LOG) is lifting your data and dumping it before the AT command handler can see it.
- ensure after each AT command a \r\n is sent. I used 'Real term' - 'Send' tab which allows a Send ascii with EndOfLine '\r' to be sent.

I have not needed to look at flow control.. if the processor is running at 240MHz it is well fast enough to pic uart data at 115200.

tcpipchip, 2019-10-10:
i will try

about 

**UART1 is for sending AT commands and receiving response, but its pins can be changed. The pins of UART1 is in the factory_param.bin, they can be changed in the esp-at/components/customized_partitions/raw_data/factory_param/factory_param_data.csv.**

Wich of those must i modify and the compiler identify  to me to define the AT commands to UART 0 , pin 3 and 1 ?

PLATFORM_ESP32,WROOM-32,0xfcfc,1,1,78,1,13,CN,115200,17,16,15,14,-1,-1
PLATFORM_ESP32,WROVER-32,0xfcfc,1,2,78,1,13,CN,115200,22,19,15,14,-1,-1
PLATFORM_ESP32,PICO-D4,0xfcfc,1,3,78,1,13,CN,115200,22,19,15,14,-1,-1
PLATFORM_ESP32,SOLO-1,0xfcfc,1,4,78,1,13,CN,115200,17,16,15,14,-1,-1
PLATFORM_ESP8266,WROOM-02,0xfcfc,1,0,78,1,13,CN,115200,15,13,3,1,-1,-1
PLATFORM_ESP8266,WROOM-5V2L,0xfcfc,1,1,78,1,13,CN,115200,15,13,3,1,5,-1

I ask you because in
osboxes@osboxes:~/esp-at/module_config$ ls
module_esp32_default  module_esp8266_default  module_wrover-32
There is no WROOM

tcpipchip, 2019-10-10:
yes
I will do some more tests!
If i change anything, default txd and rxd will be GPIO 16 and GPIO17 to AT commands and GPIO 1 and GPIO3 to log! Right ?

tcpipchip, 2019-10-10:
after 33 hours tryng i sucessfully runs the AT COMMANDS version **2.0** on NINA W102, looks the problems was in a python lib xlrd. I observed a warning when was trying to open  the CSV table.
After runs 
pip install xlrd --user
I solved my problem!
pip install xlrd --user

[12 more replies]