espressif/esp-at #76: Remote IP Address not showing up properly in second +IPD packet when using CIPDINFO

schwac, 2018-03-08:
When I used the AT+CIPDINFO=1 command, I see the remote IP show up properly in the first recvied pacaket, but not the subsequent packets. Those packets show 0.0.0.0

I think this might be a bug?

yake827, 2018-03-20:
Hi @schwac , I have tested the command ,and found it is ok. 
`AT+CWMODE=1

OK
AT+CWJAP="HUAWEI001","12345678"
WIFI CONNECTED
WIFI GOT IP

OK
AT+CIPMUX=1

OK
AT+CIPDINFO=1

OK
AT+CIPSTART=0,"TCP","192.168.3.137",8080
0,CONNECT

OK

+IPD,0,5,192.168.3.137,8080:12345
+IPD,0,5,192.168.3.137,8080:12345
+IPD,0,5,192.168.3.137,8080:12345
+IPD,0,8,192.168.3.137,8080:12345788
+IPD,0,10,192.168.3.137,8080:1234578834`

Can you post your test command?

xcguang, 2018-06-30:
Hi @schwac,
I will close the issue because of no update for a long time. If the problem persists, please reopen it again, thanks.

closed 2018-06-30 by xcguang