espressif/esp-at #77: Source code for libat_core.a

ESP32DE, 2018-03-12:
https://github.com/espressif/esp32-at/issues/26

@FayeY
only for clear statement:
is the at core at this time not open source ( cause developing the AT Core )
or is the at core never be open source in future?
can you please answere this for now and clear statement for future question in this?
thank you.

why i ask:
we are all in open source! we are community and we work all together to get the code run better.
in endproduct i would never use a closed LIB that i can not itself controll by source.
this is the reason why i use not BT / BLE and so on with ESP32.

if there is never a open source for AT LIB i would never use it in end product,
also i do not start to learn all commands, handle and so on then.

yes i know, i can feel free to start my own at core.
but this is not the question in "open source things" and using ESP32 next 15 years lifetime

this is the reason why i use not the AT core Lib for endroduct
and use not esp32 for this in endproduct.
cause if there is a mistake in code ( this was more times in the past )
we can not edit the "LIB" as soon as possible we want edit byself.
and we was waiting in the past for simple mistakes and question here long long time.
thats all

i think there are many people to want use AT Core,
but without source code - worthless and not profitable
rather pointless then to use a closed LIB
sry. but after 3 years using esp8266 and esp32 -
i allways feel like a waiting rat for salvation when it comes to code problems in libs.
we will deventively go from 2018 other ways.
and we had asked several times to reveal the code, nothing happened.

best wishes
rudi ;-)

FayeY, 2019-01-16:
Hi, 

We understand the point that you have, so sorry that AT core is not open source, it is our company strategy for now. Sorry for the inconvenience.

Thanks.

closed 2019-01-16 by FayeY

ESP32DE, 2019-01-16:
@FayeY 
after almost a year :)  ..
I will never understand the closed lib stories in opensource - but many thanks anyway for the statement. 
I close the chapter here for me now too.
All the Best!