espressif/esp-at #78: When using WPS, the SSID is not saved.

perm27, 2018-03-12:
In EPS-WROOM-32 when using WPS, the SSID is not saved.
Immediately after the WPS command is executed, SSID is set and it works well.
However, it seems that you can not connect automatically with AP when restarting or turning on / off the power again.

 EPS-WROOM-32

	AT+CWMODE=1
	OK
	AT+CWAUTOCONN=0
	OK
	AT+WPS=1
	OK
	WIFI CONNECTED
	WIFI GOT IP
	
	AT+CWJAP?
	+CWJAP:"xxxxxx","xx:xx:xx:xx:xx:xx",2,-40
	OK
	
	AT+WPS=0
	wps stoped
	OK
	AT+CWAUTOCONN=1
	OK
	
	AT+RST
	OK
	
	ready

 It seems that it is not automatically connected after AT + RST.

 EPS-WROOM-02
	
	AT+CWMODE=1
	OK
	AT+CWAUTOCONN=0
	OK
	AT+WPS=1
	wps started
	OK
	WIFI DISCONNECT
	wps success,connecting ap ...
	WIFI CONNECTED
	WIFI GOT IP
	
	AT+CWJAP?
	+CWJAP:"xxxxxx","xx:xx:xx:xx:xx:xx",2,-59
	OK
	
	AT+WPS=0
	wps stoped
	OK
	AT+CWAUTOCONN=1
	OK

	AT+RST
	OK
	...
	ready
	WIFI CONNECTED
	WIFI GOT IP
	
	AT+CWJAP?
	+CWJAP:"xxxxxx","xx:xx:xx:xx:xx:xx",2,-58
	OK

In the same operation EPS-WROOM-02 was connected.

Is there a way to save the SSID etc acquired by WPS to connect even after reboot?

xcguang, 2018-03-20:
Hi @perm27 ,
We will fix it ASAP.

xcguang, 2018-03-27:
Hi @perm27 ,
We have fixed it, please try it again.

perm27, 2018-03-27:
Hi, xcguang.
Thank you for fixing the program.
I confirmed that data can be sent to the server even after restart.

Thank you very mach.

closed 2018-03-27 by perm27