espressif/esp-at #79: Link failed when disable Bluetooth in menuconfig

Cheong2K, 2018-03-13:
I am trying to rebuild the AT firmware and I do not want to use any Bluetooth functions. I disabled the Bluetooth under menuconfig, but it failed at link time:
LD build/esp-at.elf
/Users/espressif/Desktop/GitHub/Particle/esp32-at/build/main/libmain.a(at_uart_task.o):(.literal.at_task_init+0x2c): undefined reference to `esp_bt_controller_mem_release'

xcguang, 2018-03-13:
Please disable AT ble command, then try it again. 
```make menuconfig``` --> ```Component config``` --> ```AT``` --> ```AT ble command support```.

Cheong2K, 2018-03-13:
Nope, but now it shows:
LD build/esp-at.elf
/Users/espressif/Desktop/GitHub/Particle/esp32-at/components/at/lib/libat_core.a(at_baseCmd.o):(.literal.at_queryCmdRfPower+0x4): undefined reference to `esp_ble_tx_power_get'

xcguang, 2018-03-13:
Oops,it's our carelessness, and we will resolve it ASAP. 
So sorry for the inconvenience to you.

xcguang, 2018-03-27:
@Cheong2K 
Please try it again using the latest version. If it works well for you, please update the issue.

closed 2018-04-11 by xcguang