espressif/esp-at #81: MDNS AT command missing

zavovi, 2018-03-17:
Hi all,
I downloaded the latest master from this repository and compiled. After that, I cannot use MDNS command. Is it possible enable it or how can I add it?

Thank you
Vilem

yake827, 2018-03-20:
Hi @zavovi , The current MDNS interface changes a lot between IDF2.1 and IDF3.0, and we are evaluating which set of interfaces to use, and we will reopen them as soon as possible.

zavovi, 2018-03-20:
Hi @yake827, thank you. I will wait for the new release. Could you give me an information, when it will be added?

zavovi, 2018-03-26:
Hi @yake827, is it any date, when will be new release? Thanks

xcguang, 2018-03-27:
Hi，
Please test it using the latest version
Please note we have changed AT+MDNS because of ESP-IDF API . The new command is

- AT+MDNS=\<enable>[,\<hostname>,\<service_name>,\<port>]
    - \<enable>:
        - 1: enables the MDNS function; the following three parameters need to be set.
        - 0: disables the MDNS function; the following three parameters need not to be set.
    - \<hostname>: MDNS host name
    - \<service_name>: MDNS service name
    - \<port>: MDNS port

zavovi, 2018-03-27:
Hi @xcguang,
Thank you for the new version. I tested the latest version, but the MDNS command still missing. Is it disabled in code or how can I enable it? Here is a log:

ready
AT+CIPSERVERMAXCONN=2
AT+CIPMUX=1

ready
WIFI CONNECTED
WIFI GOT IP
AT+CIPMUX=1
AT+CIPMUX=1

OK
AT+CIPSERVERMAXCONN=2
AT+CIPSERVERMAXCONN=2

OK
AT+CIPSERVER?
AT+CIPSERVER?
+CIPSERVER:0

OK
AT+CIPSERVER=1,8080
AT+CIPSERVER=1,8080

OK
AT+MDNS=1
AT+MDNS=1
ERR CODE:0x01090000

ERROR
AT+MDNS=1,1
AT+MDNS=1,1
ERR CODE:0x01090000

ERROR
AT+MDNS=1,1,"espresif","_iot",12345
AT+MDNS=1,1,"espresif","_iot",12345
ERR CODE:0x01090000

ERROR
AT+MDNS=1,"espresif","_iot",12345
AT+MDNS=1,"espresif","_iot",12345
ERR CODE:0x01090000

ERROR
AT+MDNS=0
AT+MDNS=0
ERR CODE:0x01090000

ERROR
AT+MDNS=0
AT+MDNS=0
ERR CODE:0x01090000

ERROR
AT+MDNS=0,1
AT+MDNS=0,1
ERR CODE:0x01090000

ERROR
AT+MDNS=0,1
AT+MDNS=0,1
ERR CODE:0x01090000

ERROR
AT+MDNS=1
AT+MDNS=1
ERR CODE:0x01090000

ERROR

zavovi, 2018-03-28:
Hi @xcguang, it is working now! Thank you!

xcguang, 2018-03-30:
Hi,
We have sync the latest lib at github.

closed 2018-03-30 by xcguang