espressif/esp-at #84: BLE connection between a ESP-WROOM-32 and another BLE device

mrbiondo, 2018-03-20:
Hello,
does anyone has tried to make a ble connection between a ESP-WROOM-32 and another ble device? 
If yes, how do I have to configure the characteristics' attributes in order to write to them? 
My problem is being able to write to any characteristic.
Is there anything I do wrong?
Thanks in advance for the help.

I connect a esp-wroom-32 device, to a ble device (Cypress).
the first one is central, the second one is a peripheral.
I manage to connect to the ble device, discover the primary services and characteristics, read the characteristics.
I do not manage to write to any characteristic.

All characteristics have: 
read/write permissions (no encryption, no authentication, no autorization required);
read/write properties;

following are the discovered charactetistics and their types next to. 
+BLEGATTCCHAR:"char",0,4,1,0x4CDA2F4442804C8088B111BB7F044C0D,0x0a and type uint8	
+BLEGATTCCHAR:"char",0,4,2,0xC0D6CD494D6D4831BDAC206A10E7FD68,0x0a and type sint8
+BLEGATTCCHAR:"char",0,4,3,0x9E3657E0034447AF98A601311B3B3A5C,0x0a and type uint8 array (4 bytes)
+BLEGATTCCHAR:"char",0,4,4,0xE1D42502ECD14BACB113C7F68DE00185,0x0a and type utf8s (4 bytes)

So, if I try to write any of the above characteristics, I receive a disconnection and a error message.

Example:
AT+BLEGATTCWR=0,3,4,,1
> 1
after a few seconds 
[0;32mI (2441509) BLE_AT_GATTC: EVT 4, gattc if 3
[0m
[0;32mI (2441510) BLE_AT_GATTC: EVT 41, gattc if 3
[0m
[0;32mI (2441510) BLE_AT_GATTC: EVT 5, gattc if 3
[0m
[0;32mI (2441515) BLE_AT_GATTC: DEL.1.only one node in list,free it, delete_node_from_conn_list#L1094[0m
[0;32mI (2441524) BLE_AT_GATTC: EVT 41, gattc if 4
[0m
[0;32mI (2441528) BLE_AT_GATTC: EVT 41, gattc if 5
[0m
+BLEDISCONN:0,"bb:a0:50:10:11:12"
ERROR

xiewenxiang, 2018-03-27:
Hi @mrbiondo 
Could provide all the operation steps and log?

mrbiondo, 2018-03-29:
Sorry, 
find attach the log file of the operation.
[output_2018-03-29_10-57-15.log](https://github.com/espressif/esp32-at/files/1859322/output_2018-03-29_10-57-15.log)

xiewenxiang, 2018-03-29:
@mrbiondo  Thanks, we will check it.

mrbiondo, 2018-04-04:
Hello, any news about the possible bug I reported? 
As I really need that feature, I would like to know any comment about that. 
Thanks.

xiewenxiang, 2018-04-05:
Hi @mrbiondo， 
I feel so sorry for not replying to you in time.
The attachment is the `esp-at.bin`. Please try it, if you have any questions, please let me know.

[esp-at.bin.gz](https://github.com/espressif/esp32-at/files/1880457/esp-at.bin.gz)

mrbiondo, 2018-04-06:
Hi, 
I just programmed the esp_at.bin and run the firmware, but I found a problem. 
The firmware reboots after the command AT+BLEGATTCWR completes. 
Below is what you can read from the uart0.

/Users/xiewenxiang/esp/esp-at/esp-idf/components/freertos/./queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40087728 on core 0

Backtrace: 0x4008ab64:0x3ffb3340 0x4008ac63:0x3ffb3360 0x40087728:0x3ffb3380 0x4013c6d2:0x3ffb33c0 0x4013c61a:0x3ffb33e0 0x4012fcc1:0x3ffb3400 0x4012d35f:0x3ffb3470 0x40144969:0x3ffb34a0 0x40142226:0x3ffb34e0

Rebooting...

I also would like to configure the firmware by using memuconfig. Will that be possible?
Thanks.

xiewenxiang, 2018-04-08:
Hi @mrbiondo ,
[esp-at.tar.gz](https://github.com/espressif/esp32-at/files/1886975/esp-at.tar.gz)

This `esp-at.bin` will  be work fine.

mrbiondo, 2018-04-09:
Hi,
the problem I reported still exists. 
The mistake is in using the "Write Request/Response ATT operation" when the property of the characteristic is "write no respose". Instead, you must use  "Write command ATT operation". 
check it out, please.

xiewenxiang, 2018-04-09:
Hi, I have check it just now.
The client:
```
AT+BLECONN=0,"24:0a:c4:02:10:0e"

OK
+BLECONN:0,"24:0a:c4:02:10:0e"
AT+BLEGATTCPRIMSRV=0
+BLEGATTCPRIMSRV:0,1,0x1801,1
+BLEGATTCPRIMSRV:0,2,0x1800,1
+BLEGATTCPRIMSRV:0,3,0xA002,1

OK
AT+BLEGATTCCHAR=0,3
+BLEGATTCCHAR:"char",0,3,1,0xC300,0x02
+BLEGATTCCHAR:"desc",0,3,1,1,0x2901
+BLEGATTCCHAR:"char",0,3,2,0xC301,0x04
+BLEGATTCCHAR:"desc",0,3,2,1,0x2901
+BLEGATTCCHAR:"char",0,3,3,0xC302,0x08
+BLEGATTCCHAR:"desc",0,3,3,1,0x2901
+BLEGATTCCHAR:"char",0,3,4,0xC303,0x04
+BLEGATTCCHAR:"desc",0,3,4,1,0x2901
+BLEGATTCCHAR:"char",0,3,5,0xC304,0x08
+BLEGATTCCHAR:"char",0,3,6,0xC305,0x10
+BLEGATTCCHAR:"desc",0,3,6,1,0x2902
+BLEGATTCCHAR:"char",0,3,7,0xC306,0x20
+BLEGATTCCHAR:"desc",0,3,7,1,0x2902
+BLEGATTCCHAR:"char",0,3,8,0xC307,0x02
+BLEGATTCCHAR:"desc",0,3,8,1,0x2901

OK
AT+BLEGATTCWR=0,3,4,,1
>
OK
```
The server:
```
+BLECONN:0,"24:0a:c4:01:4d:ca"
AT+BLEGATTSSRV?
+BLEGATTSSRV:1,1,0xA002,1

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300,0x02
+BLEGATTSCHAR:"desc",1,1,1,0x2901
+BLEGATTSCHAR:"char",1,2,0xC301,0x04
+BLEGATTSCHAR:"desc",1,2,1,0x2901
+BLEGATTSCHAR:"char",1,3,0xC302,0x08
+BLEGATTSCHAR:"desc",1,3,1,0x2901
+BLEGATTSCHAR:"char",1,4,0xC303,0x04
+BLEGATTSCHAR:"desc",1,4,1,0x2901
+BLEGATTSCHAR:"char",1,5,0xC304,0x08
+BLEGATTSCHAR:"char",1,6,0xC305,0x10
+BLEGATTSCHAR:"desc",1,6,1,0x2902
+BLEGATTSCHAR:"char",1,7,0xC306,0x20
+BLEGATTSCHAR:"desc",1,7,1,0x2902
+BLEGATTSCHAR:"char",1,8,0xC307,0x02
+BLEGATTSCHAR:"desc",1,8,1,0x2901

OK
+WRITE:0,1,4,,1,a
```
The characteristic 0xC303 is "write without response".

mrbiondo, 2018-04-16:
Hi,
sorry for the delay. 
I spent some time investigating the issue, exchanging roles (central and peripheral) and devices(Esp32, smartphone, Cypress). I still do not understand why Cypress as peripheral does not work well with Esp32 as central when sending a "Write without responce".

Anyway, what you send me in the last message is clear. 
I'd better buy some sniffer tool to help me working out problems. 
What kind of tool do you usually use? Is that cheap? Or what tool do you recommend?
Thanks.

xiewenxiang, 2018-04-17:
Hi,
I was used the packet sniffer[http://www.ti.com/tool/packet-sniffer](url)

xcguang, 2018-06-28:
Hi @mrbiondo,
I will close this issue because of no update for a long time, if have any question, please reopen it again, thanks.

closed 2018-06-28 by xcguang