espressif/esp-at #85: Removing these log during the start up of the AT firmware

Cheong2K, 2018-03-21:
I do not want to show the log during the AT firmware startup. Is it possible to disable them?

Broets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4576
load:0x40078000,len:11640
load:0x40080000,len:896
entry 0x400802b8
I (29) boot: Detected ESP32
I (48) boot: ESP-IDF v2.1-65-g8bca703 2nd stage bootloader
I (49) boot: compile time 07:53:05
I (49) boot: Enabling RNG early entropy source...
I (66) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : DIO
I (91) boot: SPI Flash Size : 4MB
I (104) boot: Partition Table:
I (115) boot: ## Label            Usage          Type ST Offset   Length
I (138) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (161) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (184) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (208) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (231) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (254) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (277) boot: End of partition table
I (291) boot: Disabling RNG early entropy source...
I (307) boot: Loading app partition at offset 00100000
I (1424) boot: segment 0: paddr=0x00100018 vaddr=0x3f400020 size=0x1c09c (114844) map
I (1425) boot: segment 1: paddr=0x0011c0bc vaddr=0x3ffc0000 size=0x02cec ( 11500) load
I (1447) boot: segment 2: paddr=0x0011edb0 vaddr=0x40080000 size=0x00400 (  1024) load
I (1469) boot: segment 3: paddr=0x0011f1b8 vaddr=0x40080400 size=0x00e50 (  3664) load
I (1497) boot: segment 4: paddr=0x00120010 vaddr=0x400d0018 size=0x8ec7c (584828) map
I (1521) boot: segment 5: paddr=0x001aec94 vaddr=0x40081250 size=0x10f4c ( 69452) load
I (1581) boot: segment 6: paddr=0x001bfbe8 vaddr=0x400c0000 size=0x00064 (   100) load
I (514) cpu_start: Pro cpu up.
I (515) cpu_start: Single core mode
I (520) heap_init: Initializing. RAM available for dynamic allocation:
I (526) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (532) heap_init: At 3FFCBCA0 len 00014360 (80 KiB): DRAM
I (539) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (545) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (551) heap_init: At 4009219C len 0000DE64 (55 KiB): IRAM
I (558) cpu_start: Pro cpu start user code
I (643) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0

ready

xcguang, 2018-03-24:
Please refer to http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html?highlight=%20log

xcguang, 2018-06-08:
Hi,
Does it work well?  I will close this issue, if the problem persists, please reopen it ,thanks.

closed 2018-06-08 by xcguang