espressif/esp-at #86: AT command response with same string

jinjunzh, 2018-03-27:
Hi Guys,
when i send command like "AT" , it received the same string like "AT", rather than "OK". I think that proves uart read/write do works ,but seems libat_core.a failed to work. do you any  suggestion for further debugging?
---------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5616
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13716
entry 0x40078fb4
\0x1b[0;32mI (30) boot: ESP-IDF v3.0-rc1-161-g8856cc0 2nd stage bootloader\0x1b[0m
\0x1b[0;32mI (30) boot: compile time 15:38:37\0x1b[0m
\0x1b[0;32mI (34) boot: Enabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (36) boot: SPI Speed      : 40MHz\0x1b[0m
\0x1b[0;32mI (40) boot: SPI Mode       : DIO\0x1b[0m
\0x1b[0;32mI (44) boot: SPI Flash Size : 4MB\0x1b[0m
\0x1b[0;32mI (48) boot: Partition Table:\0x1b[0m
\0x1b[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length\0x1b[0m
\0x1b[0;32mI (59) boot:  0 phy_init         RF data          01 01 0000f000 00001000\0x1b[0m
\0x1b[0;32mI (66) boot:  1 otadata          OTA data         01 00 00010000 00002000\0x1b[0m
\0x1b[0;32mI (74) boot:  2 nvs              WiFi data        01 02 00012000 0000e000\0x1b[0m
\0x1b[0;32mI (81) boot:  3 at_customize     unknown          40 00 00020000 000e0000\0x1b[0m
\0x1b[0;32mI (89) boot:  4 ota_0            OTA app          00 10 00100000 00180000\0x1b[0m
\0x1b[0;32mI (96) boot:  5 ota_1            OTA app          00 11 00280000 00180000\0x1b[0m
\0x1b[0;32mI (104) boot: End of partition table\0x1b[0m
\0x1b[0;32mI (108) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x21138 (135480) map\0x1b[0m
\0x1b[0;32mI (164) esp_image: segment 1: paddr=0x00121160 vaddr=0x3ffc0000 size=0x03358 ( 13144) load\0x1b[0m
\0x1b[0;32mI (169) esp_image: segment 2: paddr=0x001244c0 vaddr=0x40080000 size=0x00400 (  1024) load\0x1b[0m
\0x1b[0;32mI (171) esp_image: segment 3: paddr=0x001248c8 vaddr=0x40080400 size=0x0b748 ( 46920) load\0x1b[0m
\0x1b[0;32mI (199) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xd3200 (864768) map\0x1b[0m
\0x1b[0;32mI (499) esp_image: segment 5: paddr=0x00203220 vaddr=0x4008bb48 size=0x096e8 ( 38632) load\0x1b[0m
\0x1b[0;32mI (516) esp_image: segment 6: paddr=0x0020c910 vaddr=0x400c0000 size=0x00064 (   100) load\0x1b[0m
\0x1b[0;32mI (528) boot: Loaded app from partition at offset 0x100000\0x1b[0m
\0x1b[0;32mI (528) boot: Disabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (529) cpu_start: Pro cpu up.\0x1b[0m
\0x1b[0;32mI (532) cpu_start: Single core mode\0x1b[0m
\0x1b[0;32mI (537) heap_init: Initializing. RAM available for dynamic allocation:\0x1b[0m
\0x1b[0;32mI (544) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (550) heap_init: At 3FFD37B0 len 0000C850 (50 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (556) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (562) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (569) heap_init: At 40095230 len 0000ADD0 (43 KiB): IRAM\0x1b[0m
\0x1b[0;32mI (575) cpu_start: Pro cpu start user code\0x1b[0m
\0x1b[0;32mI (593) cpu_start: Starting scheduler on PRO CPU.\0x1b[0m
\0x1b[0;32mI (664) uart: queue free spaces: 10\0x1b[0m
I (668) wifi: wifi firmware version: e6ab128
I (668) wifi: config NVS flash: enabled
I (670) wifi: config nano formating: disabled
\0x1b[0;32mI (674) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE\0x1b[0m
\0x1b[0;32mI (684) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE\0x1b[0m
I (706) wifi: Init dynamic tx buffer num: 32
I (707) wifi: Init data frame dynamic rx buffer num: 32
I (707) wifi: Init management frame dynamic rx buffer num: 32
I (710) wifi: wifi driver task: 3ffdefe0, prio:23, stack:4096
I (716) wifi: Init static rx buffer num: 10
I (719) wifi: Init dynamic rx buffer num: 32
I (724) wifi: wifi power manager task: 0x3ffe369c prio: 21 stack: 2560
\0x1b[0;32mI (762) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0\0x1b[0m
I (764) wifi: mode : softAP (24:0a:c4:09:8f:69)
I (771) wifi: mode : sta (24:0a:c4:09:8f:68) + softAP (24:0a:c4:09:8f:69)
I (774) wifi: mode : softAP (24:0a:c4:09:8f:69)

ready

AT
AT
AT
-----------------------------

MaJerle, 2018-03-27:
Send CRLF characters after AT and you will see the magic.

AT\r\n

jinjunzh, 2018-03-28:
It  output the same string as before. I also add log into function of at_port_read_data & at_port_write_data. When I send “AT\r\n”, it returns as below:
------------LOG--------------------
[ZJJDBG]at_port_read_data buf[0]=,len=6
[ZJJDBG]at_port_write_data data[0]=A,len=6
AT\r\n

jinjunzh, 2018-03-28:
Hi Guys,
AT works now. Thanks you so much

closed 2018-03-28 by jinjunzh