espressif/esp-at #87: GATT Service Bin Creation

mrbiondo, 2018-04-05:
Hello,
Is it possible to define more than one service with the procedure described in the toos folder?
If yes, can anyone explain how to do that?
Or, is there an alternative way to define and flash more than one service?
It seems that AT commands, like AT+BLEGATTSSRVSTART and AT+BLEGATTSSRVSTOP, can manage more than one service independently.
Thanks for any comment.

xiewenxiang, 2018-06-29:
Hi ,
Only one service is currently supported.

kurta999, 2018-08-10:
Any news on this issue? When it will be supported? Our company would need it, this ESP32 chip would be the best solution for us.

xcguang, 2020-03-15:
Can you try commit b991a183bd6645340fc73a07cc964cfb3c5aae99?

xiewenxiang, 2020-03-19:
This feature is supported, Please use the latest commit

closed 2020-03-19 by xiewenxiang