espressif/esp-at #88: How to use at Bluetooth to connect to the phone？

FlyingArm, 2018-04-16:
I used the AT firmware library and also initialized the Bluetooth related section. However, no Bluetooth device can be found on the phone, resulting in an inability to connect to the esp32 board.
So if you want to connect esp32 Bluetooth to your phone, how do you do it?

xiewenxiang, 2018-04-17:
Hi,
Maybe you can refer to this document.
[esp32_at_instruction_set_and_examples_en.pdf](https://github.com/espressif/esp32-at/files/1918293/esp32_at_instruction_set_and_examples_en.pdf)

FlyingArm, 2018-04-17:
I have read the document,and i set the ble to be a client but the phone can't to search it .

xiewenxiang, 2018-04-17:
You can try the following steps.
```
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADVSTART

OK
```
Now turn on your phone, the app which I was used is `LightBlue`.

FlyingArm, 2018-04-17:
My phone is Android ,thank you i will test later!

FlyingArm, 2018-04-17:
But AT+BLEINIT=2 it will set BLE be server ? Why ?

xiewenxiang, 2018-04-17:
Because most of the phones are BLE client

xiewenxiang, 2018-04-17:
I your phone is Android , you can use the APP `BLEDeng`

xcguang, 2018-06-15:
Hi,
I will close this issue, if it still persists, please reopen it, thanks.

closed 2018-06-15 by xcguang