espressif/esp-at #89: busy p...

zavovi, 2018-04-17:
Hello,
how is it working the "busy p..."? 
When I tried send lot of bytes, I sometimes get "busy p...". But after that I received "Recv XXX", where XXX is count of the bytes, which I want to send. After that I received "SEND OK".

Why I received "Recv XXX" and "SEND OK"? The "busy p..." tells me, that the data cannot be send, is it really true?

Thank you for the clarification.

xcguang, 2018-04-18:
Maybe the "\r\n" or other character is following your data.

zavovi, 2018-04-18:
Hi @xcguang, 
oh, but I send the binary data (e.g. picture in binary stream). Is it bad? Because ther could be "\r\n" like data, not like end of the stream. Could you help me, how can I solve it? 

Is AT commands only for send text, or can I send the byte stream?
Thanks.

xcguang, 2018-04-18:
The binary data is OK. 
If you test AT command using serial tools in PC, please disable new line mode in the tools settings when sending data, and enable it when sending command.
I guess the busy you encountered is because you enable new line mode in the tool settings, which will append the "\r\n" to your data. 
Or you can monitor the serial line to make sure all the data are sent by you.

zavovi, 2018-04-18:
I am not using no tool in PC. I have the code in C, which is sending the data over UART. I made something like socket API on the AT commands and now, I am tuning and cleaning issues. 

I will sniff the data on the UART and on TCP. 

Thanks.

xcguang, 2018-06-15:
Hi,
Is the issue resolved?  I will close it, and if the problem still exists, please reopen it, thanks.

closed 2018-06-15 by xcguang