espressif/esp-at #90: AT Command Does not Work

zackgd1024, 2018-04-24:
I can compile and flash the esp32-at code with make flash under Linux.
However, after esp32 boot, it does not respond to any AT Command.
Here is my log ,
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
[0;32mI (28) boot: ESP-IDF v3.0-rc1-278-gba13d282 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 23:38:29[0m
[0;32mI (28) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (38) boot: SPI Mode       : DIO[0m
[0;32mI (42) boot: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (65) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (79) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (87) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (102) boot: End of partition table[0m
[0;32mI (106) boot: No factory image, trying OTA 0[0m
[0;32mI (111) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20768 (132968) map[0m
[0;32mI (167) esp_image: segment 1: paddr=0x00120790 vaddr=0x3ffc0000 size=0x03214 ( 12820) load[0m
[0;32mI (172) esp_image: segment 2: paddr=0x001239ac vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (174) esp_image: segment 3: paddr=0x00123db4 vaddr=0x40080400 size=0x0c25c ( 49756) load[0m
[0;32mI (203) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xd8eb0 (888496) map[0m
[0;32mI (514) esp_image: segment 5: paddr=0x00208ed0 vaddr=0x4008c65c size=0x09514 ( 38164) load[0m
[0;32mI (531) esp_image: segment 6: paddr=0x002123ec vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (543) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (543) boot: Disabling RNG early entropy source...[0m
I (660) wifi: wifi firmware version: 6450435
I (661) wifi: config NVS flash: enabled
I (661) wifi: config nano formating: disabled
I (670) wifi: Init dynamic tx buffer num: 32
I (670) wifi: Init data frame dynamic rx buffer num: 32
I (670) wifi: Init management frame dynamic rx buffer num: 32
I (676) wifi: wifi driver task: 3ffdf5d8, prio:23, stack:4096
I (681) wifi: Init static rx buffer num: 10
I (685) wifi: Init dynamic rx buffer num: 32
I (689) wifi: wifi power manager task: 0x3ffe3ce4 prio: 21 stack: 2560
I (719) wifi: mode : softAP (30:ae:a4:19:90:61)
I (725) wifi: mode : sta (30:ae:a4:19:90:60) + softAP (30:ae:a4:19:90:61)
I (729) wifi: mode : softAP (30:ae:a4:19:90:61)

Has anyone met with this problem?
Thanks

zackgd1024, 2018-04-24:
I have tried the “Hello World” esp32-idf example. And it works fine.

dcsuperman, 2018-04-24:
I met the same issue...
It seems to be a failure to boot

xcguang, 2018-04-24:
Hi，
AT port is uart1, please refer to "Hardware Introduction"  in ```https://github.com/espressif/esp32-at/blob/master/README.md.```

zackgd1024, 2018-04-25:
@xcguang Thanks. 
After change to use GPIO16 as RXD and GPIO17 as TXD. AT command works fine.

closed 2018-04-25 by zackgd1024