espressif/esp-at #91: AT and event_handler

santanapablo1975, 2018-04-24:
Hi All,

I am trying to test the event handler after calling the AT lib but it is not workin some how... here thstatic esp_err_t event_handler(void *ctx, system_event_t *event);
**********
```
**extern void at_task_init(void);
void app_main(){
    at_task_init();    
    esp_error_check = esp_event_loop_init(event_handler, NULL);
    if (esp_error_check != ESP_OK) {
    	printf("printf: no ESP_OK \r\n");
    } else {  
    	printf("printf: ESP_OK \r\n");
    }
}

static esp_err_t event_handler(void *ctx, system_event_t *event){
    switch(event->event_id) {
    case SYSTEM_EVENT_ETH_START:
    	at_write_port_main="\r\n SYSTEM_EVENT_ETH_START \r\n";
        printf("printf: %s ", at_write_port_main);        break;
    default:
    	break;
    }
    return ESP_OK;
}**
```

here the console

```
I (2421) wifi: mode : softAP (30:ae:a4:59:e8:81)
printf: no ESP_OK 
```

ready

ON the other had i I decide to uncomment the call to the     at_task_init(); it will work...

```
void app_main(){
    //at_task_init();    
    esp_error_check = esp_event_loop_init(event_handler, NULL);
    if (esp_error_check != ESP_OK) {
    	printf("printf: no ESP_OK \r\n");
    } else {  
    	printf("printf: ESP_OK \r\n");
    }
}
```
any body would know why is this behavior?

here the logs with out at_task_init

```
I (1221) cpu_start: Starting scheduler on PRO CPU.
printf: ESP_OK 
```

Thanks

Regards

xcguang, 2018-04-25:
Hi，
esp_event_loop_init  has been called in at_task_init. And we will add some commands about ethernet next version.

santanapablo1975, 2018-04-25:
Ok thanks for the update!

closed 2018-04-25 by santanapablo1975