espressif/esp-at #94: Classic Bluetooth AT commands?

cwq9999, 2018-05-14:
I have been successful in setting up the BLE operation; however, I was hoping to be able to run in dual mode BLE/BT classic.  I notice that I can turn on the classic bluetooth option in menuconfig, however, I don't see any AT commands for classic bluetooth operations and when I activate the classic bluetooth in the menuconfig when I scan for bluetooth devices in Android nothing appears.

xiewenxiang, 2018-05-17:
Classic bluetooth command is not supported in the current version.

xcguang, 2018-06-30:
Hi @cwq9999 ,
We have added some AT commands about classic bluetooth on the master branch, please refer to https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Commands_Set.md

subbash19, 2018-07-21:
Dear @xcguang,

Thanks for the AT commands for classic bluetooth. The AT commands are not updated in the AT Commands pdf file on Espressif and git hub link mentioned above.  Does these AT commands work with older version....?. 

Thanks in advance

xcguang, 2018-07-21:
Hi @subbash19 ,
Now these commands about classic bluetooth just work in the master branch, the commands in the pdf are released version, and these classic bluetooth commands have not been released. Maybe they will be released in release/v1.3.

subbash19, 2018-07-21:
Dear @xcguang,

When can we expect the release of v1.3.

We have the esp32 dev kit. We wanted to test the BT classic functionality.
Can you suggest any other ways to test the BT classic functionality if the
v1.3 code release will be delayed.

Thanks and Regards,
Subbash

xcguang, 2018-07-21:
Hi @subbash19 ,
You can clone the master branch, then compile the AT binary. And I'm not sure current functions satisfy you, just feel free to let us know if you have any advice. 
Please watch the project, and you will receive email if the project have any update, in the future, we will create release/v1.3 based on master branch.

xcguang, 2018-09-29:
Hi,
Now the master branch supports classic bluetooth commands, please try it again.
I will close the issue, if you have any question, please reopen it again, thanks.

closed 2018-09-29 by xcguang

techbatman29, 2020-07-17:
Hi @xcguang,
I am not able to access this link: https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Commands_Set.md anymore.
Moreover, The Espressif AT Commands pdf is yet to have updated on Bluetooth Classic AT Commands!!.. If you have any resources for Bluetooth Classic AT Commands for ESP32 please share them here..

Thank you in advance

xcguang, 2020-07-17:
@techbatman29  you can get it from
[image]

techbatman29, 2020-07-17:
> @techbatman29 you can get it from
> [image]

Thank You @xcguang