espressif/esp-at #95: esp32 Flash using ESP32 DOWNLOAD TOOL Not working ,after fflash

senthil313, 2018-05-15:
hi,
i develop the code using arduino ide...
i generate .bin file for programming...
i used ESP32 DOWNLOAD TOOL ...for flashing our device...
after flash, it reset continuously....
kindly give any other solution for upload code .bin or .hex  format...
[image]
[image]

senthil313, 2018-05-15:
i change the flash QIO for bootloader_qio.bin and DIO for bootloader.bin...
i change this into reverse also...

[image]

xcguang, 2018-05-22:
Please change SPI SPEED 40MHz，and try it. Thanks.

xcguang, 2018-06-19:
Hi,
Is the issue resolved? I will close it because of no update from you for a long time, and you can reopen it if the problem persists. Thanks for your interest in esp32-at.

closed 2018-06-19 by xcguang

senthil313, 2018-06-19:
Hi xcguang,
Yhanks for your reply...
now its working.... follow the Verbose and clear those problems...

kapyaar, 2019-06-09:
@senthil313 Hi, I am using arduino to generate the bin file and spiffs binary. Do you know the addresses that should be used in this case?