espressif/esp-at #100: Error message during make

pwpolar, 2018-06-07:
esp/esp32-at/main/interface/uart/at_uart_task.c:41:20: fatal error: esp_bt.h: No such file or directory
compilation terminated.

This file exists at esp/esp32-at/esp-idf/components/bt/include.

Why am I getting this error message?

xcguang, 2018-06-08:
Hi,
Please remove sdkconfig in the project directory, then `make defconfig` and `make` to compile the project.

pwpolar, 2018-06-08:
Thank you for your reply,

Unfortunately still get same error message.

xcguang, 2018-06-09:
Hi,
Which commit id of the project you use? And please provide your sdkconfig.

pwpolar, 2018-06-10:
Thank you for your help.

Commit ID : bf7fc41e

xiewenxiang, 2018-06-12:
Hi pwpolar,

Does this commit is right?
I can't find this commit in both esp-at project and esp-idf project.

pwpolar, 2018-06-13:
Took esp-at/esp-idf and copied it to esp/esp-idf.
Ran make
Here is result:
fatal: Not a git repository: /home/pwooster/esp/esp-idf/../.git/modules/esp-idf
fatal: Not a git repository: /home/pwooster/esp/esp-idf/../.git/modules/esp-idf
WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

xiewenxiang, 2018-06-14:
```
cd ~
git clone https://github.com/espressif/esp32-at.git
cd esp32-at
git submodule update --init --recursive
export IDF_PATH=~/esp32-at/esp-idf
make
```

pwpolar, 2018-06-15:
Thanks for your help. Issue resolved. Moving esp32-at directory to root helped (originally was in esp directory.

closed 2018-06-15 by pwpolar