espressif/esp-at #101: busy p... to any AT command

JuPrgn, 2018-06-11:
Hi all,

I am using an ESP32-WROOM devkit and would like to send AT commands from PUTTY. But I need some help please.

Configuration : 

I am using esp32-at latest master commit : 713bbecbc59f45fb2bc5fe072c209785deee4342.

I use make menuconfig to configure AT UART to use the same USB-UART on the devkit used for flashing and configure it as follow : 
UART module 0
UART_Rx : GPIO3
UART_Tx : GPIO1
UART_RTS : GPIO13
UART_CTS : GPIO15

I use the default msys2 environnement : esp32_win32_msys2_environment_and_toolchain-20180110

Issue : 

I successfully flash the firmware and get "ready" from ESP32, then I write : 
**AT+RST** (pressing ENTER, CTRL+J which correspond to LF on putty, and again ENTER)

And ESP32 respond with : 
**busy p...**
**ERR CODE:0x010b0000**

After multiples try I found that if I write : 
**AT+RST** (pressing CTRL+J which correspond to LF on putty, and ENTER)
followed by : 
**AT+RST** (pressing ENTER, CTRL+J which correspond to LF on putty, and again ENTER)

The command is working and ESP reset.

This only works for this command RST, any other command respond with busy p error.

Let me know if you need more informations. Thank you for your help !

Please find below the booting output : 

`$ make monitor
MONITOR
--- idf_monitor on COM25 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5808
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:14960
entry 0x4007862c
I (30) boot: ESP-IDF v3.1-dev-1193-g64b56bef-dirty 2nd stage bootloader
I (30) boot: compile time 13:27:00
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (75) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (82) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (90) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (97) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (105) boot: End of partition table
I (109) boot: No factory image, trying OTA 0
I (114) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1fd18 (130328) map
I (169) esp_image: segment 1: paddr=0x0011fd40 vaddr=0x3ffc0000 size=0x002d0 (   720) load
I (169) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0xe8ee4 (954084) map
0x400d0018: _flash_cache_start at ??:?

I (509) esp_image: segment 3: paddr=0x00208f04 vaddr=0x3ffc02d0 size=0x03644 ( 13892) load
I (515) esp_image: segment 4: paddr=0x0020c550 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Julien/Logiciel/ESP32/Firmware/esp32-at/esp-idf/components/freertos/xtensa_vectors.S:1685

I (517) esp_image: segment 5: paddr=0x0020c958 vaddr=0x40080400 size=0x15048 ( 86088) load
I (561) esp_image: segment 6: paddr=0x002219a8 vaddr=0x400c0000 size=0x00064 (   100) load
I (574) boot: Loaded app from partition at offset 0x100000
I (574) boot: Disabling RNG early entropy source...

ready
`

xcguang, 2018-06-11:
Hi, 
Maybe the https://github.com/espressif/esp32-at/issues/51 issue can help you.

JuPrgn, 2018-06-11:
Hi @xcguang ,

Thank you for your fast answer, I already seen this issue and tried the 3 options CR, LF, and CR LF but I get similar result : 

**busy p...
ERR CODE:0x010b0000

busy p...
ERR CODE:0x010b0000

...
**

JuPrgn, 2018-06-11:
@xcguang,

I was also thinking that using the same UART for AT and console could be a problem and desactivated UART console but I got this same error. 
Do you know if I must use a second USB to Serial converter for AT UART or if I can use at the same time the one used for flashing and debug ?

xcguang, 2018-06-12:
Hi,
You can use the flashing port for AT, but you need configure AT port by `make menuconfig`.
Could you give all log you reproduce? and maybe you can try other serial tool, for example, cutecom in linux.

JuPrgn, 2018-06-12:
Hi,

Thank you very much ! You have right this is working with pqcom serial tool !
I don't know what is going wrong with my Putty config/usage...

Best regards

xcguang, 2018-06-13:
Sorry for that I can't give you helpful suggestions because I didn't use Putty.

JuPrgn, 2018-06-13:
No problems I will experiment and let you know if I found something usefull.

JuPrgn, 2018-06-13:
Putty experiments : 

Configuration:
Serial tab :  115200 baud, 8 data bits, 1 stop, no parity, XON/XOFF Flow control.
Terminal tab : Local echo = force on, local line editing, force on

Putty always send CR when pressing ENTER.
To send LF with Putty you have to press CTRL+J (which write ^J in terminal) and press ENTER to send 
 which then send LF+CR. **It seems impossible to send CR+LF.**

ESP experiments : 

With make menuconfig : Configuring Line ending for UART input (**CR**) 

I tested some commands but it seems ESP is always waiting for CR+LF at startup for the first command. If you send only CR, it will echo commands but you can't get any response. 
After receiving CR+LF, the ESP will answer to all the commands terminated by CR or terminated by CR+LF.

**Please find below my experiments**

Step 1 at startup :
 
PC to ESP : 
« AT+GMR » with CR
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D

ESP to PC : 
« AT+GMR » with CR
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D

**Echo ACK command but no response to the command.**

Step 2 :

PC to ESP : 
« AT+GMR » with CR-LF
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D 0x0A

ESP to PC : 
« AT+GMR » with CR-LF
« ERR CODE:0x01090000

ERROR »
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D 0x0A
0x45 0x52 0x52 0x20 0x43 0x4F 0x44 0x45 0x3A ... 0x0D 0x0A

**ACK command + Error « previous command + current command » it seems ESP was waiting for CR-LF to finish the first command.**

 
Step 3 :

PC to ESP : 
« AT+GMR » with CR-LF
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D 0x0A

ESP to PC : 
« AT+GMR » with CR-LF
« AT version:1.1.0.0(7d44009 - Jun 1 2018 11:09:12)
SDK version:v3.1-dev-1193-g64b56bef-dirty
compile time:Jun 13 2018 13:19:06

OK »
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D 0x0A
0x41 0x54 0x20 0x76 0x65 0x72 0x73 0x69 ... 0x0D 0x0A

**ACK command + response to the command = SUCCESS**

Step 4 :

PC to ESP : 
« AT+GMR » with only CR
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D

ESP to PC : 
« AT+GMR » with CR-LF
« AT version:1.1.0.0(7d44009 - Jun 1 2018 11:09:12)
SDK version:v3.1-dev-1193-g64b56bef-dirty
compile time:Jun 13 2018 13:19:06

OK »
0x41 0x54 0x2B 0x47 0x4D 0x52 0x0D 0x0A
0x41 0x54 0x20 0x76 0x65 0x72 0x73 0x69 ... 0x0D 0x0A

**ACK command + response to the command SUCCESS**

xcguang, 2018-06-15:
Hi,
How do you configure Line ending for AT command terminator?
If you use `CR` for AT command terminator, please set it like this.
```
[*]   AT command terminator support.  
   (0x0D)  AT command terminator.
```

JuPrgn, 2018-06-15:
Hi,

Thank you for your great support ! 
I was not configuring it in the right menu **AT** but in **ESP32-specific** "Line ending for UART input"...

closed 2018-06-18 by JuPrgn