espressif/esp-at #103: AT+CIUPDATE doesnt work

mdelgadom, 2018-06-19:
Hello, I have flash Espressif binary file 1.0.0.0. Once my ESP-WROOM-32 is connected to the Internet, I have problems with the command AT+CIPUPDATE

AT+CIUPDATE
+CIPUPDATE:1
+CIPUPDATE:2
+CIPUPDATE:3

ERROR

In debug console I see:

I (42233) event: ip: 192.168.101.11, mask: 255.255.254.0, gw: 192.168.100.254
ota_mode:0
http request length 389 bytes
version:v1.1.0.0
total_len=1113312!
OTA Write Header format Check Failed! first byte is 00

What is wrong?

I want to update from 1.0.0.0 to 1.1.0.0, using your cloud, like esp32_at_instruction_set_and_examples_en.pdf says:

AT+CIUPDATE—Updates the Software Through Wi-Fi
If using Espressif's AT BIN (/esp-idf/bin/at), AT+CIUPDATE will download a new AT BIN from the
Espressif Cloud.

I have flashed these files:

[image]

xcguang, 2018-06-22:
Hi,
I test it, and it works well. Please try the release bin at https://www.espressif.com/en/support/download/at, thanks.
Following is the upgrade log ,
```
[2018-06-22_12:36:58:846]ready
[2018-06-22_12:37:02:176]AT+GMR
[2018-06-22_12:37:02:176]AT version:1.0.0.0(870285c - Nov 14 2017 07:26:38)
[2018-06-22_12:37:02:176]SDK version:v2.1-65-g8bca703
[2018-06-22_12:37:02:176]compile time:Nov 17 2017 07:53:25
[2018-06-22_12:37:02:176]Bin version(Wroom32):1.0.0
[2018-06-22_12:37:02:176]OK
[2018-06-22_12:37:08:988]AT+CWMODE=1
[2018-06-22_12:37:09:004]
[2018-06-22_12:37:09:004]OK
[2018-06-22_12:37:11:317]AT+CWJAP="HUAWEI001","espressif"
[2018-06-22_12:37:13:071]WIFI CONNECTED
[2018-06-22_12:37:13:757]WIFI GOT IP
[2018-06-22_12:37:14:316]
[2018-06-22_12:37:14:316]OK
[2018-06-22_12:37:33:394]AT+CIupdate
[2018-06-22_12:37:33:505]+CIPUPDATE:1
[2018-06-22_12:37:33:537]+CIPUPDATE:2
[2018-06-22_12:37:33:649]+CIPUPDATE:3
[2018-06-22_12:38:14:870]+CIPUPDATE:4

[2018-06-22_12:38:14:870]OK
[2018-06-22_12:38:14:870]WIFI DISCONNECT
[2018-06-22_12:38:17:024]
[2018-06-22_12:38:17:024]ready
[2018-06-22_12:38:17:166]WIFI CONNECTED
[2018-06-22_12:38:17:916]WIFI GOT IP
[2018-06-22_12:38:19:363]AT+GMR
[2018-06-22_12:38:19:363]AT version:1.1.0.0(e83e1f4 - Jun  1 2018 07:10:27)
[2018-06-22_12:38:19:363]SDK version:v3.0.1-rc
[2018-06-22_12:38:19:363]compile time:Jun 11 2018 12:42:49
[2018-06-22_12:38:19:363]Bin version(Wroom32):1.1.0
[2018-06-22_12:38:19:363]OK

```

mdelgadom, 2018-06-22:
After your comment, I have noticed that the problem was with my firewall/traffic inspector. Once I've disabled that now it's working!. Thanks you very much.

I close the issue.

closed 2018-06-22 by mdelgadom