espressif/esp-at #104: cannot make successfully!

ldsheng, 2018-06-22:
I have cloned the whole project include the submodules use the "git clone --recursive https://github.com/espressif/esp32-at.git" command, and then follow the compiling guide:

rm sdkconfig
make defconfig
make

at the end of build, I met this error:

...
CC build/wpa_supplicant/src/fast_crypto/fast_sha256-internal.o
CC build/wpa_supplicant/src/fast_crypto/fast_aes-wrap.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa-debug-module/trax.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/esp-at.elf
D:/work/EADB/code/Firmware/ESP32/esp32-at/components/at/lib\libat_core.a(at_port.o):(.literal.initialise_wifi+0x14): undefined reference to `g_wifi_osi_funcs'
collect2.exe: error: ld returned 1 exit status
make: *** [D:\work\EPCID2IN1\WirelessBoard\Firmware\epcid-esp32-firmware\esp-idf/make/project.mk:406：/d/work/EADB/code/Firmware/ESP32/esp32-at/build/esp-at.elf] 错误 1

does any body build the  libat_core.a version c3c7f6fb sucessfully?

JuPrgn, 2018-06-22:
Hi,
Sometimes I get error using "make" command and using "make all" solve this maybe you can try ?

ldsheng, 2018-06-23:
@JuPrgn Thx for the response. I have tried the "make all" command and got same error.

xcguang, 2018-06-27:
Hi @ldsheng ,
Please try `rm -rf build`, and then `make`.

ldsheng, 2018-06-28:
HI @xcguang ,
Have tried that. remove the whole build directory and make again. It still reflects same error. Exactly same as I posted at the beginning.

yake827, 2018-06-28:
Hi @ldsheng  ' g_wifi_osi_funcs ' is defined on  esp-idf/components/esp32/wifi_os_adapter.c L400, can you find the API under your ESP-IDF?   You can renew the esp-at project and submodule by `git submodule update --init --recursive`  and then `export IDF_PATH=/path of esp-at/esp-idf/` to export the IDF

ldsheng, 2018-06-28:
It's OK now. 

Thanks @yake827 , thank you for your information.
It turns out that I have several version on my disk, and I forgot to re-export the IDF path to the one I recently work on:(. I have found the defination of the g_wifi_osi_funcs on wifi_os_adapter.c L400 and then I realized it's a version mismatch problem. And then I checked the IDF_PATH and BINGO! :) Thank you guys!

closed 2018-06-28 by xcguang