espressif/esp-at #105: ESP32-AT get crushed when been sent AT with 2 "\r\n"

ldsheng, 2018-06-23:
I‘m using the libat_core.a version e83e1f46 from the branch release/v1.1.0.0.

I found if I send the command "AT\r\n" the  ESP32 response correctly. However, if I send the command "AT\r\n\r\n" which is a "AT" with 2 return and newline charactors, the system get curshed  and it will  keep on sends "ERR CODE:0x010b0000 busy p..." untill I reset it.

It looks like this:

AT

OK

AT
ERR CODE:0x010b0000

busy p...
ERR CODE:0x010b0000

busy p...
ERR CODE:0x010b0000

busy p...
ERR CODE:0x010b0000

busy p...

I have also tried the AT+GMR command, it acts same way.

xcguang, 2018-06-27:
Hi @ldsheng ,
Yes, I have reproduced it, and we will fix it as soon as possible. If any update, I will reply here, thanks very much.

xcguang, 2018-06-30:
Hi @ldsheng ,
We have fix the issue, please try again using the latest master branch or release/v1.1.0.0.
Now we are testing the release/v1.1.0.0 branch, if everything is ok, we will release binary file and tag v1.1.1.0 onto release/v1.1.0.0 branch.

ldsheng, 2018-07-02:
Hi @xcguang ,

I have tested the last commit. It's fixed now. Thanks for your hard work. Good job!

closed 2018-07-02 by ldsheng

GIYura, 2019-06-21:
Hi, i work with esp32-wroom module. Sometimes the module response on the commant AT+CIPSTART= ...\r\n with the error ERR CODE:0x010b0000 busy p...it happens from time to time.