espressif/esp-at #106: No regular TCP connection closed

zjykymt, 2018-06-25:
I tested the stability of the ESP32 connection TCP, and then often the situation that TCP was closed.
I enabled the TCP keep-alive function and the detection time is 10s.

# Log duration for each TCP connection
[image]

# ESP32 Tx pin monitoring log
[image]

# My application log
[image]

xcguang, 2018-06-27:
Hi, 
Could you provide some packets? for example, the WiFi packets captured by the omnipeek or the ethernet packets of the server by the wireshark?

zjykymt, 2018-07-09:
Currently I can't capture this type of packet here, I will close this ISSUE first, if I confirm the problem will open.

zjykymt, 2018-09-07:
Hi,

At present, we have decided to put the ESP32 module into production, but this problem still exists. This problem makes my team feel not very stable.

What I want to know is how many cases ESP32 modules will send 0, CLOSE.

AT version is 1.1.2.0

Thanks.

xcguang, 2018-09-07:
Could you provide some packets? for example, the WiFi packets captured by the omnipeek or the ethernet packets of the server by the wireshark?

zjykymt, 2018-09-07:
I am using wireshark to capture the package.
Once this happens, I will upload the packet capture data.

zjykymt, 2018-09-07:
[image]

Hi, this is one of the cases, I will always monitor for more packet capture data.

zjykymt, 2018-09-07:
[image]
[image]

This is the other two TCP disconnected packet capture data.

zjykymt, 2018-09-07:
Hi,

It seems that the client is actively resetting TCP.
Can these data help determine the cause?

xcguang, 2018-09-30:
I guess it maybe ESP32 does not receive the packet from the air for a long time, or ESP32 sends data to the server because of the server does not receive them, then ESP32 closes the socket. When receiving the data again,  ESP32 will reset the peer because the socket has been closed.

closed 2019-01-16 by FayeY