espressif/esp-at #107: ESP32-PICO-D4 - AT commands and UART conflict with Flash

WanaGo, 2018-07-04:
Hello,

We are trying to use the ESP32-PICO-D4 and were told to use the AT BIN files for the WROVER, which should work on the PICO. We had tried the WROOM-32 but that did not boot, but the WROVER ones are working OK.

The next question is the UART that the AT commands use. On the WROOM-32 development board, the AT commands come out of GPIO16 and GPIO17, which is said to be UART1. 
On the PICO-D4, it states that GPIO16 and GPIO17 cannot be used, as they are used for the Internal Flash, which is shown in the Schematic in the datasheet.
So how do you use the AT Commands if the UART the AT commands use, is reserved for the Flash Memory chip?

Can the AT commands UART be changed to UART0, like the ESP8266 was, so everything goes out the one UART? If this is possible, how do you do it?

In the md file for this REPO, it states:

> The ESP32 Board sends AT commands through UART1 by default.
> GPIO16 is RXD
> GPIO17 is TXD

In the PICO Datasheet though, **GPIO16 and GPIO17 is UART2**, not UART1. U2RXD is pin 25, U2TXD is pin 27 (GPIO16 and GPIO17)
UART1 is SD2 and SD3 (pins 28 and 29, U1RXD and U1TXD), which are for the Flash memory also.

So this is confusing.

How do you use the AT command set on the PICO-D4 if the UART it defaults to is reserved for the Flash memory?
How do you change the AT command set UART to be on UART0 instead?

Thank you.

WanaGo, 2018-07-04:
I have seen in the menuconfig, the ability to change the UART now, however would this be correct?
[image]

I tried doing a build yesterday, and loading that into the PICO, but it failed to boot, as though the build made it suit the WROOM-32 by default, not the WROVER/PICO. 
How do you change the build to suit the PICO?

I cant find documentation on this stuff, for someone who has never done this, its not obvious.

WanaGo, 2018-07-04:
OK 40 and 41 above are not correct. I was assuming they are Pin number, not GPIO number.

However, how do you customise the build to run on a PICO, as the built will not run, it however will run on a WROOM-32.
What needs to change for the PICO ?

WanaGo, 2018-07-04:
And what GPIO is default in the AT BIN files for WROVER, for the AT UART?
If the WROVER build is the one to use for PICO, where does the UART end up on, what GPIO?

WanaGo, 2018-07-04:
OK thank you @xcguang 

I will try that now.

Do you know how to compile the build of BIN files from source, for the WROVER32 or PICO, instead of the WROOM-32 ?
What other changes need to be made?
I cannot find documentation on this.

Thanks

xcguang, 2018-07-04:
Hi，
After configuring UART pin, you just need run `make` to compile. It will output the bin files in the build directory.Or run `make flash` to download the bin files into the flash.

WanaGo, 2018-07-04:
Changed the UART pints for AT, to GPIO3 and GPIO1, to try and have AT commands working on UART0 along with console messages, like the ESP8266 did.
Changed it in the menuconfig, and disabled flow control by setting it to 0.
Went to build using MAKE, and got this:

james@THEBEAST MSYS ~/esp/esp32-at
$ make
C:/msys32/home/james/esp/esp32-at/esp-idf/make/project.mk:60: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
GENCONFIG
/home/james/esp/esp32-at/esp-idf/make/project.mk:60: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
/bin/sh: python: command not found
make[1]: *** [/home/james/esp/esp32-at/esp-idf/components/esptool_py/Makefile.projbuild:51: /home/james/esp/esp32-at/build/bootloader/bootloader.bin] Error 127
make: *** [/home/james/esp/esp32-at/esp-idf/components/bootloader/Makefile.projbuild:41: /home/james/esp/esp32-at/build/bootloader/bootloader.bin] Error 2

Deleted the esp32-at directory, and started again, same result.
It worked fine before I changed the AT UART...
What am I doing wrong?

xcguang, 2018-07-04:
Hi,
The error is `/bin/sh: python: command not found`, please ensure python has been installed.

WanaGo, 2018-07-04:
There is nothing in the instructions which talks about python, or what to install, where to install etc. Is there instructions I am missing somewhere?
I do already have Python 3.6 64-bit installed from a previous project, is this sufficient?

JuPrgn, 2018-07-04:
Hi
I had to install latest python and set add to path environnement variable and then I got a new error with a missing submodule I can't remember the name but doing some research I found the package and installed it in python. After these step this was working perfectly. 
Let me know if this is working for you !

WanaGo, 2018-07-04:
hmm.
> james@THEBEAST MINGW32 ~/esp/esp32-at
> $ which python
> /mingw32/bin/python

So python is already there as part of this.

I then did a make, but still had the error.
I then tried make all. After some time, it completed, and it had generated esp-at.bin, phy_init_data.bin and partitions_at.bin. I then tried make again, and i didnt get the error.

Quite a lot of errors in the make all though.

Man, this is not straightforward for someone who has never done this.
Why are the instructions not complete?

I then uploaded the esp-at.bin to the PICO-KIT and now the UART is UART0, which is great, but why is the build having so many problems after following the instructions on this repo?
Can the instructions be updated so it is complete?

WanaGo, 2018-07-04:
james@THEBEAST MINGW32 ~
$ cd ~/esp

james@THEBEAST MINGW32 ~/esp
$ cd esp32-at

james@THEBEAST MINGW32 ~/esp/esp32-at
$ git pull
Already up to date.

james@THEBEAST MINGW32 ~/esp/esp32-at
$ git submodule update

james@THEBEAST MINGW32 ~/esp/esp32-at
$ cd esp-idf

james@THEBEAST MINGW32 ~/esp/esp32-at/esp-idf
$ git submodule update

james@THEBEAST MINGW32 ~/esp/esp32-at/esp-idf
$ cd ..

james@THEBEAST MINGW32 ~/esp/esp32-at
$ rm sdkconfig

james@THEBEAST MINGW32 ~/esp/esp32-at
$ make defconfig
DEFCONFIG
#
# configuration written to /home/james/esp/esp32-at/sdkconfig
#
GENCONFIG

james@THEBEAST MINGW32 ~/esp/esp32-at
$ make menuconfig
MENUCONFIG
configuration written to /home/james/esp/esp32-at/sdkconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

james@THEBEAST MINGW32 ~/esp/esp32-at
$ make
GENCONFIG
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/cpu_util.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
[... 916 lines cut ...]
CC build/wpa_supplicant/src/wps/wps.o
CC build/wpa_supplicant/src/wps/wps_dev_attr.o
CC build/wpa_supplicant/src/wps/wps_attr_parse.o
CC build/wpa_supplicant/src/wps/wps_validate.o
CC build/wpa_supplicant/src/wps/wps_attr_process.o
CC build/wpa_supplicant/src/wps/eap_common.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa-debug-module/trax.o
CC build/xtensa-debug-module/eri.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/esp-at.elf
esptool.py v2.3.1
CC phy_init_data.o
BIN phy_init_data.bin
To flash all build output, run 'make flash' or:
python /home/james/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/james/esp/esp32-at/build/bootloader/bootloader.bin 0x20000 /home/james/esp/esp32-at/build/at_customize.bin 0xf000 /home/james/esp/esp32-at/build/phy_init_data.bin 0x100000 /home/james/esp/esp32-at/build/esp-at.bin 0x8000 /home/james/esp/esp32-at/build/partitions_at.bin

james@THEBEAST MINGW32 ~/esp/esp32-at
$ make
To flash all build output, run 'make flash' or:
python /home/james/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/james/esp/esp32-at/build/bootloader/bootloader.bin 0x20000 /home/james/esp/esp32-at/build/at_customize.bin 0xf000 /home/james/esp/esp32-at/build/phy_init_data.bin 0x100000 /home/james/esp/esp32-at/build/esp-at.bin 0x8000 /home/james/esp/esp32-at/build/partitions_at.bin

james@THEBEAST MINGW32 ~/esp/esp32-at
$ which python
/mingw32/bin/python

WanaGo, 2018-07-04:
Found this in another Issue post (this needs to be in the instructions, surely...)

C:\msys32\mingw32\bin>python -m pip install -U pip
Collecting pip
  Using cached https://files.pythonhosted.org/packages/0f/74/ecd13431bcc456ed390b44c8a6e917c1820365cbebcb6a8974d1cd045ab4/pip-10.0.1-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Uninstalling pip-9.0.1:
      Successfully uninstalled pip-9.0.1
Successfully installed pip-10.0.1

C:\msys32\mingw32\bin>pip install pyyaml xlrd
Collecting pyyaml
  Downloading https://files.pythonhosted.org/packages/4a/85/db5a2df477072b2902b0eb892feb37d88ac635d36245a72a6a69b23b383a/PyYAML-3.12.tar.gz (253kB)
    100% |################################| 256kB 2.8MB/s
Collecting xlrd
  Downloading https://files.pythonhosted.org/packages/07/e6/e95c4eec6221bfd8528bcc4ea252a850bffcc4be88ebc367e23a1a84b0bb/xlrd-1.1.0-py2.py3-none-any.whl (108kB)
    100% |################################| 112kB 5.1MB/s
Installing collected packages: pyyaml, xlrd
  Running setup.py install for pyyaml ... done
Successfully installed pyyaml-3.12 xlrd-1.1.0

Make now builds successfully.

xcguang, 2018-07-05:
Hi,
About the developing environment for idf, please refer to https://github.com/espressif/esp-idf/blob/master/README.md , which is a link in esp-at project README. 
But I think the description about the python packages is missing in README, and it is https://github.com/espressif/esp32-at/blob/master/tools/README.md

tchouzin, 2018-08-06:
Hello #WanaGo,

Did you find a solution for the AT command? Were you able to execute AT commands after you configure the module after builds?

Thank you for your answer.

WanaGo, 2018-08-06:
Hi @tchouzin 
Yes it works, after stumbling along from broken information, I did get it working as per the above.

tchouzin, 2018-08-07:
Hello @WanaGo and thank you for your answer,
I followed the steps you mentioned above. I manage to build the code without error, but I can not test AT commands, as mentioned below:

https://github.com/espressif/esp32-at/issues/117

I do not know what else to do. I did different methods but without success when I entered AT commands

closed 2018-08-07 by xcguang

galyamovm, 2019-07-15:
> Hi @tchouzin
> Yes it works, after stumbling along from broken information, I did get it working as per the above.

Hello, WanaGo! Can you post your esp-at.bin file?

Fosyan, 2020-02-22:
GJ!!! thanks for all your effort