espressif/esp-at #108: ESP32-AT Rebooting Error

tcli, 2018-07-07:
Hi, 
I flashed the latest esp32-at firmware yesterday, and had the same issue with [@DIMOOLLA ](https://github.com/espressif/esp32-at/issues/99) which saying that "ota data partition invalid and no factory, will try all partitions."
I had tried to erase the flash by using "make erase_flash", and then "make flash" to flash the AT firmware to ESP32. I used "make monitor" to see the debug log of the esp32, but I saw the ESP32 now keeps rebooting and giving me the same message like the attached picture:
[image]

P.S I did one change in "menuconfig", which is the <Default serial port> under the <Serial flasher config>.

Any advise or guidance would be greatly appreciated!
Thanks,
TC

tcli, 2018-07-07:
P.S I did one change in "menuconfig", which is the <Default serial port> under the <Serial flasher config>.

JuPrgn, 2018-07-08:
Hi I had the same issue and I found a blank file to load to erase all memory ans this worked for me. There is an  esp tool to load bin files at different address.

tcli, 2018-07-08:
I was wondering what address should I load the blank bin file to? Should I erase the flash first, and then load the blank bin file to ESP32? After all these, then flash the AT firmware into the ESP? Thanks

JuPrgn, 2018-07-08:
I loaded it at address 0 and erased the entire memory. And then I flashed again AT firmware. I think I had some old datas on memory that were not erased with erase command. If you load the blank file with the flash download tool it will overwrite your AT firmware partitions but I don't remember if there is a erase button on this tool.

JuPrgn, 2018-07-08:
This may help http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/.

tcli, 2018-07-08:
Awesome! Thanks for all the info. 
I will try to load the blank file to 0x00000 and also the address for AT firmware!

JuPrgn, 2018-07-08:
You are welcome ! Please let me know if this is working for you good luck ! 😉

tcli, 2018-07-10:
Hi JuPrgn,
I tried to use the official flash download tool to flash the [ESP3232-WROOM-32 AT Bin V1.1](https://www.espressif.com/en/support/download/at) into my ESP32, but this time the ESP froze at "wifi mode" (attached picture).
[image]
I tried to send "AT", but the ESP32 won't return "OK" as well as other AT commands. 
I was wondering do you have any ideas about this issue?
Thanks!

JuPrgn, 2018-07-10:
Hi

I didn't succeed using precompiled binary but as you don't know the config I recommend that you make flash with your own configuration. If you erased all memory this should works now.

tcli, 2018-07-11:
Hi JuPrn,
I tried to flash the AT firmware with my configuration, and still having the same rebooting issue...

JuPrgn, 2018-07-11:
Hi
That's a bad new, so you do not have the same issue, I don't know what you could try now... 
Maybe @xcguang will have some better ideas ?

xcguang, 2018-07-11:
@tcli ,
Which uart port you use to send AT command? Is the uart pin correct? Could you try the default configuration?

tcli, 2018-07-12:
@xcguang @JuPrgn 
I was still having the reboot issue when flashing the AT firmware that was cloned from Github. Fortunately, I was just able to get the AT firmware that downloads directly from Espressif site working....

xcguang, 2018-08-07:
@tcli Could you provide your sdkconfig?

xcguang, 2018-10-11:
I will close the issue because of no update for a long time.

closed 2018-10-11 by xcguang