espressif/esp-at #109: "busy p..." appears when two AT commands are sent continuously, after the module will always be in the state of sending "busy p..."

zjykymt, 2018-07-09:
I tried to send the two commands "ATE0" and "AT+GMR" consecutively after receiving "ready", and "busy p..." appears , after the module will always be in the state of sending "busy p..."

Version:
AT version:1.1.0.0(e83e1f4 - Jun  1 2018 07:10:27)
SDK version:v3.0.1-rc
compile time:Jun 11 2018 12:42:49
Bin version(Wroom3{15:12:00:221}2):1.1.0

[image]

xcguang, 2018-07-19:
Please try the latest release/v1.1.0.0.  We have fix it.
https://github.com/espressif/esp32-at/commits/release/v1.1.0.0

zjykymt, 2018-07-23:
The latest release/V1.1.1.0 have fixed it. 

Thanks.

closed 2018-07-23 by zjykymt