espressif/esp-at #110: Classic Bluetooth Initialization Error

ChM4rio, 2018-07-13:
Hi,
I've got an Espressif ESP32-PICO-Kit v4. I successfully added the esp32-AT Firmware and enabled the "AT bt command support" in the menuconfig, in order to use bt classic SPP. I also changed the AT Uart Settings, disabled the UART for console output and set the optimization level to "Release" in menuconfig. The Firmware should be the latest version from git (pulled 3 days ago, 10.07.2018).

I can query the AT+BTINIT State without problem:
```
AT+BTINIT?
+BTINIT:0
OK
```
But when I try to initialize the Bluetooth Classic, I get an error (without error number):
```
OK
AT+BTINIT=1
ERROR
```
Before the AT+BTINIT Command I resetted the device with AT+RST and made sure that BLE is not initialized (AT+BLEINIT? returned 0) and that WiFi ist not enabled (AT+CWMODE returned 0).

Is there something else I need to configure or check, before I can initialize the BT-Classic? Is this a know issue and/or is BT-Classic not supported yet?

Any advise or guidance would be appreciated
Greetings
Mario

JuPrgn, 2018-07-18:
Hi 
I have a similar issue with Err Code 0x01090000 and 0x01080000 for some commands and my ESP restart during AT+BTSTARTDISC command...
I had a memory problem when building project I had to disable a lot of functionnality to avoid : "region `dram0_0_seg' overflowed" did you had similar messages ?

xiewenxiang, 2018-07-20:
@ChM4rio @JuPrgn 
Hi,
The problem is there is not enough memory, the ESP32-PICO-Kit v4 don't have PSRAM.
If you have `ESP-WROVER-KIT`,  `ESP32-LyraTD-MSC` or `ESP32-LyraT` , you can enable the menuconfig,

```
Components config->AT->AT bt command support.
Components config->Bluetooth->Bluedroid Enable->BT/BLE will first malloc the memory from the PSRAM 
Components config->Bluetooth->Bluedroid Enable->Use dynamic memory allocation in BT/BLE stack
```

ChM4rio, 2018-07-20:
I had no time to test it nor to check if I had similar memory problems when building the project yet. But I will try it out next week and then I'll tell you.

But anyway thanks for the replies/input.
Greetings

ChM4rio, 2018-07-25:
@JuPrgn I rebuild (and flashed) the project and the same message was shown in the console window:
"region 'dram0_0_set' overflowed by 6080 bytes", but only when I disable "dynamic memory alloation in BT/BLE stack".

Then I tried to disable some features, but no luck yet. 

But for the final product, we need another chip (ESP32-WROOM32D) anyway, so I stop trying to get BT-SPP to work with the PICO.
Issue closed.

closed 2018-07-30 by ChM4rio

SamDhoundiyal, 2018-08-21:
Hi,
i am using ESP32-wroom with 520 KB of on-chip SRAM for data and instructions and 
448 KB of ROM for booting and core functions. I want to execute BT Classic with set AT Commands 

Then I tried to disable some features. and try to config classic. but esp32 restart after this many command and i will receive ready in terminal as receiver when power on module with at-command.

ready
AT+SYSRAM?
+SYSRAM:101168
OK

AT+BTINIT=1
OK

AT+SYSRAM?
+SYSRAM:35076
OK

AT+BTSTARTDISC=0,10,1
OK

ready

Any advise or guidance would be appreciated,

Regards 
SAM

SamDhoundiyal, 2018-08-21:
new issues created.

https://github.com/espressif/esp32-at/issues/123