espressif/esp-at #111: Classic bluetooth Reset

JuPrgn, 2018-07-18:
Hi,

I am using the latest ESP32-at (2018-07-18) and trying to use Classic BT but I get a reboot when starting AT+BTSTARTDISC. 

Please let me know if you need more informations/testing.

I attach a log of my AT commands : 

[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (65) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (73) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (80) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (88) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (95) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (103) boot: End of partition table[0m
[0;32mI (107) boot: No factory image, trying OTA 0[0m
[0;32mI (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x23a74 (146036) map[0m
[0;32mI (172) esp_image: segment 1: paddr=0x00123a9c vaddr=0x3ffc0000 size=0x03844 ( 14404) load[0m
[0;32mI (178) esp_image: segment 2: paddr=0x001272e8 vaddr=0x3ffc3844 size=0x00000 (     0) load[0m
[0;32mI (179) esp_image: segment 3: paddr=0x001272f0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (188) esp_image: segment 4: paddr=0x001276f8 vaddr=0x40080400 size=0x08918 ( 35096) load[0m
[0;32mI (211) esp_image: segment 5: paddr=0x00130018 vaddr=0x400d0018 size=0x10bce4 (1096932) map[0m
[0;32mI (596) esp_image: segment 6: paddr=0x0023bd04 vaddr=0x40088d18 size=0x0d448 ( 54344) load[0m
[0;32mI (619) esp_image: segment 7: paddr=0x00249154 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (619) esp_image: segment 8: paddr=0x002491c0 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (638) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (638) boot: Disabling RNG early entropy source...[0m

ready
AT+BTINIT=1

OK
AT+BTSCANMODE=2

OK
AT+BTSTARTDISC=0,10,1
at_bt_gap_cb - event = 1

OK
at_bt_gap_cb - event = 0

Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x401d72ce  PS      : 0x00060330  A0      : 0x800f33ce  A1      : 0x3ffea790  
A2      : 0x4000c458  A3      : 0x00000009  A4      : 0x3ffea7d0  A5      : 0x3f4001b4  
A6      : 0x00000002  A7      : 0x3ffea664  A8      : 0x00000000  A9      : 0x00000002  
A10     : 0x3fff8c94  A11     : 0x3ffea622  A12     : 0x3fff8c96  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4000c458  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x401d72ce:0x3ffea790 0x400f33cb:0x3ffea7b0 0x400e888c:0x3ffea7d0 0x400e8a11:0x3ffea800 0x400f922d:0x3ffea840 0x400f940e:0x3ffea990 0x400f6e12:0x3ffea9b0

Rebooting...
ets Jun  8 2016 00:22:57

JuPrgn, 2018-07-18:
I also tried to read and set unsuccessfully BT name : 

AT+BTNAME?
+BTNAME:
OK

AT+BTNAME="esp"
ERROR

JuPrgn, 2018-07-19:
Switching to release mode change error : 

> AT+BTSTARTDISC=0,10,1
> at_bt_gap_cb - event = 1
> 
> 
> OK
> at_bt_gap_cb - event = 0
> 
> Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x401c29be  PS      : 0x00060f30  A0      : 0x800f22f6  A1      : 0x3ffea7a0  
> A2      : 0x4000c3e4  A3      : 0x00000009  A4      : 0x3ffea7e0  A5      : 0x3f4001b4  
> A6      : 0x00000002  A7      : 0x3ffea674  A8      : 0x00000000  A9      : 0x00000002  
> A10     : 0x3fff8c20  A11     : 0x3ffea632  A12     : 0x3fff8c22  A13     : 0x3ffd7c10  
> A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000016  EXCCAUSE: 0x00000003  
> EXCVADDR: 0x4000c3e4  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x401c29be:0x3ffea7a0 0x400f22f3:0x3ffea7c0 0x400e79a0:0x3ffea7e0 0x400e7b25:0x3ffea810 0x400f70d5:0x3ffea850 0x400f74bd:0x3ffea870 0x400f764a:0x3ffea9c0 0x400f583e:0x3ffea9e0
> 
> Rebooting...
> ets Jun  8 2016 00:22:57

JuPrgn, 2018-07-19:
I am not sure that I should use BT Classic for my project I will be more precise about what I am trying to achieve : 
I want to connect a Dualshock 4 (PS4) gamepad to my esp32 and cannot figure out if this BLE or BT classic. The chip is a Qualcomm AR3002 which is a Bluetooth 4.0 device (BLE?) but it seems this is using SDP and L2CAP (I am a bluetooth newbie). 
I found this hack that is reading data transfered from DS4 : http://www.psdevwiki.com/ps4/DS4-BT.
Anyway I cannot find if this device is using BLE or BT and when I try to scan I see none on BLE and can not try BT Classic (which reboot during scan).
Any help would be apreciated

xiewenxiang, 2018-07-20:
@JuPrgn 
Currently, there are a lot of exceptions when using Classic Bluetooth related commands, because memory is almost completely used up, so, the BT command is disable by default. We are now working on optimize memory.

JuPrgn, 2018-07-23:
@xiewenxiang 
Thank you very much ! 
Have you planned to implement it in the coming months or it is not planned at the moment ?

xcguang, 2019-11-12:
Hi, 
we have added ble hid command, I will close this issue, if you have any question, please reopen it again.

closed 2019-11-12 by xcguang

JuPrgn, 2019-11-12:
Thank you for this update. I will give it a try

reshma26p, 2019-11-13:
i got following response when bluetooth of PC connected to esp32  BT..
not getting device index and device address 
_bt_gap_cb - event = 4