espressif/esp-at #112: Tools BLEService

JuPrgn, 2018-07-20:
Hi 
I am using python 3.6.5 and BLEService.py is not working on python 3.x.

I don't know how to do a pull request, when I try to open a new pull request I can only compare branch but I never pulled before so I probably do it the wrong way.

I have a local clone from your repo with a new branch and a commit that fix the followings errors : 

Fix error using python 3.x :

- Add parenthesis for print function
- except standarderror, e -> except Exception as e
- python3 "unicode" doesn't exist anymore but all string are unicode -> str

Script is still not working I get :

esp32-at\tools>python BLEService.py GattService
Example.csv
b'index'

And there is no BIN output file on the folder.

I can put the script content here or try to pull request if you give some explanations =)

xcguang, 2019-11-12:
Please try it again, my python version is `Python 3.8.0`, it works well.

xcguang, 2019-11-12:
I will close this issue firstly, if have any question, please reopen it again.

closed 2019-11-12 by xcguang