espressif/esp-at #117: Can't enter any ESP32 AT-commands on Windows using any terminal (putty, tera term ....)

tchouzin, 2018-08-01:
Hello,
I hope can get your help and I appreciate you in advance.

Can anybody tell me an simple procedure to setup and build AT-command on Windows from scratch? 

I followed the procedure described in the following link without success: https://github.com/espressif/esp32-at 

I also looked at the "issues" section, but could not solve the problem.

Any response would be very helpful.
Thanks

Achille T.

[image]
[image]

JuPrgn, 2018-08-01:
Hi you have to configure the UART your are using to send at commands using menuconfig depending on your hardware you will chose UARTx and you will configure the gpio pins. You can also configure \r or \n or both as end of command I had some issue using Putty an I used pqcom on Windows with that terminal you can select which ending you want but don't forget to check the checkbox on this menu that doesn't work if this is not checked.

tchouzin, 2018-08-05:
Thanks for your answer #JuPrgn. 
I configured the gpio pins and UART and checked that everything was correct. But I still have no expected results.
I have two modules esp32 ( https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview and https://www.espressif.com/en/products/hardware/esp32-devkitc/overview ). But I have the same results.

The modules restart in infinite loop when I type the command "make monitor", or when i open the port in another terminal. 

I really need help.

Thank you in advance for giving me some of your time to help me make it happen, because I need it for a school project.

Here are some screenshots of configurations
[image]
[image]

tchouzin, 2018-08-06:
Hello everyone,
I keep looking for a solution to my problem specified above. Despite the configurations that are made, I still do not receive anything. I wonder what I missed in this step that seems clear.
I also used the pqcom terminal as suggested by #JuPrgn, but that did not solve the problem.

I ask once again for your help.

Thank you in advance.

Below, some screenshots.
[image]

[image]

tchouzin, 2018-08-07:
Hello @xcguang 
Any help on this?

Thanks

xcguang, 2018-08-07:
Hi @tchouzin ，
I guess the serial tools do not send “\r\n", please try this tool. 

[image]

[QCOM_V1.1.zip](https://github.com/espressif/esp32-at/files/2266459/QCOM_V1.1.zip)

tchouzin, 2018-08-07:
Hello @xcguang ,
Thank you for your help. It works now with the QCOM terminal.

Thanks again.

closed 2018-08-07 by xcguang

rkashish75, 2022-09-22:
Can anyone help with AT commands through Arduino IDE for ESP32?
https://github.com/xcguang can you help?

xcguang, 2022-09-22:
Hi @rkashish75 What we need to do? We can evaluate it。

rkashish75, 2022-09-22:
I need to scan continuously for BT devices in vicinity through enquiry scan
and send the MAC address, of the detected devices to a web server
periodically.
We did this using Arduino Mega, ESP01 Wi-Fi and BT HC-05 in master mode.
BT is used for continuous enquiry scan.

xcguang, 2022-09-22:
Hi, 
Now do you want to replace Arduino Mega, ESP01 Wi-Fi and BT HC-05 with ESP32?
We have not used Arduino, can you provide more detail about the issue?

rkashish75, 2022-09-22:
Yes. I want to replace the combined hardware with ESP-32.
My problem is that AT commands do not work as smooth and simple as in case
of Arduino Uno/Mega and BT HC-05 combination.
I have installed the ESP_AT Library by Khoi Hoang.
Also did the complete toolchain and tried using the mingw33.exe.
Didn't get the expected response to AT commands. Whereas I tried QCOM V1.1
as suggested by you. Port opens and I get the initial booting information
and few devices are discovered by default.

I need to do continous scanning for nearby BT devices which are
discoverable. I don't need any pairing or handshake. Only enquiry.
Send the collected Mac I'ds periodically to a web server.