espressif/esp-at #120: AT+CIPSEND not working after browser page connects to TCP server on ESP32.

Quaqmire, 2018-08-15:
I have followed the example in the ESP32 AT Instruction Set and Examples from pages 73-75 for setting up a TCP server.   I can set up the TCP server, connect to the network with my laptop, and I can request a connection from a browser (Chrome 68.0.3440.106 running on Windows 7) to IP address 192.168.4.1.   The ESP32 receives the request and reports a connection (1,CONNECT).

However, any variation of the AT+CIPSEND command that I enter after that only gives me "ERROR", and I cannot send data back to the browser.  

Any suggestions or advice to help me resolve this would be much appreciated!  Thanks in advance.

Below are the AT commands that I entered via a terminal emulator:

AT+RESTORE
1,CLOSED
0,CLOSED

OK

ready
AT+GMR
AT version:1.2.0.0-dev(c3c7f6f - Jun 20 2018 07:40:00)
SDK version:v3.1-dev-1381-g9a55b42f
compile time:Jun 26 2018 14:02:43

OK
AT+CWSAP="MyESP32SSID","MyPassword",5,3,5,0

OK
AT+CWMODE=3

OK
AT+CIPMUX=1

OK
AT+CIPSERVER=1,80

OK
AT+CIPSTO=1000

OK
AT+CIPDINFO=1

OK
AT+CWLIF

OK

+IPD,0,405,192.168.4.2,49519:GET / HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/68.0.3440.106 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

0,CONNECT
1,CONNECT
AT+CWLIF
+CWLIF:192.168.4.2,xx:xx:xx:xx:xx:xx

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","192.168.4.2",49519,80,1
+CIPSTATUS:1,"TCP","192.168.4.2",49520,80,1

OK
AT+CIFSR
+CIFSR:APIP,"192.168.4.1"
+CIFSR:APMAC,"xx:xx:xx:xx:xx:xx"
+CIFSR:STAIP,"0.0.0.0"
+CIFSR:STAMAC,"xx:xx:xx:xx:xx:xx"

OK
AT+CIPSEND=0,4

ERROR
AT+CIPSENDEX=0,4

ERROR
AT+CIPSEND=4

ERROR
AT+CIPSENDEX=4

ERROR
AT+CIPSEND=1,4,"192.168.4.2",49520

ERROR
AT+CIPSENDEX=1,4,"192.168.4.2",49520

ERROR
AT+CIPSEND=0,4,"192.168.4.2",49519

ERROR
AT+CIPSENDEX=0,4,"192.168.4.2",49519

ERROR
AT+CIPSEND=1,4

ERROR
AT+CIPSENDEX=1,4

ERROR
1,CLOSED

NuclearManD, 2019-04-27:
I am having a similar problem:

```AT+CIPSEND=7
ERROR

AT+CEER
+CEER: No Cause

CLOSED```

As you can see, it closes after I try AT+CIPSEND, and there is no error cause indicated.  I don't even get the '>' prompt.  What's wierd is that it works right before this too, then it just stops working.

xcguang, 2020-01-09:
Could you try the latest master branch? The the issue persists, please reopen it.

closed 2020-01-09 by xcguang