espressif/esp-at #121: at-http-webserver error

costaud, 2018-08-19:
I got this after running make flash for at-http-server example:

```
using pdrv=0f_mount failed (13)Formatting FATFS partitionMounting againesphttpd: active and listening to connections.
Httpd init
Conn req from  192.168.4.2:24292, using pool slot 0
URL = /
Is url index 0
Pool slot 0 is done. Cleaning up for next req
URL = /login.html
Is url index 34
E (196568) ParseAt: command execution error
Cannot open file:/login.html
/login.html not found. 404!
Pool slot 0 is done. Cleaning up for next req
Pool slot 0: socket closed.
```

yake827, 2018-08-20:
It seems file system mount failed,  you can  survey the build log， make sure file system image build success.

xcguang, 2018-09-29:
Is the issue resolved?

xcguang, 2018-10-29:
Close it because of no update for a long time.

closed 2018-10-29 by xcguang