espressif/esp-at #122: Running at-http-webserver and at-core at once

Wihan, 2018-08-20:
How can I use the AT-core through the UART but still use the AT-webserver, I don't want to run all the AT commands through the webserver, I'm having trouble using both.

> [22:51:47:638] ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400e9554
[22:51:47:638] file: "/builds/rd/esp-at-core/components/at_core/at_port.c" line 694
[22:51:47:638] func: initialise_wifi
[22:51:47:638] expression: esp_event_loop_init(at_wifi_event_handler_cb, NULL)
[22:51:47:638] 
[22:51:47:638] Backtrace: 0x4008facb:0x3ffbe230 0x4008ffd8:0x3ffbe250 0x400e9554:0x3ffbe270 0x400ea01f:0x3ffbe330 0x400f3a98:0x3ffbe370 0x400f1b9e:0x3ffbe3b0 0x400d2de2:0x3ffbe450 0x400d1837:0x3ffbe480

I added all the relevant cgi commands at the top of the given app_main() 

I added the initialization code at the bottom:

...
#ifdef CONFIG_AT_EAP_COMMAND_SUPPORT
    if(esp_at_eap_cmd_regist() == false) {
        printf("regist eap cmd fail\r\n");
    }
#endif

#ifdef CONFIG_AT_COMMAND_TERMINATOR
    esp_at_custom_cmd_line_terminator_set((uint8_t*)&cmd_terminator);
#endif

    ESP_ERROR_CHECK(esp_at_init_parse());

#ifndef ESP_AT_FATFS_COMMAND
    ret =  espFsInit();
    assert(ret == 1);
#endif
    httpdInit(builtInUrls, 80);

    at_custom_init();
}

yake827, 2018-08-21:
Hi  @Wihan , Currently we do not support simultaneous run.

closed 2018-08-21 by Wihan

Wihan, 2018-08-22:
@yake827 do you plan to do this in the future? The only way I can configure the ssid and password seems to be via the smart-config mode and run time via at-commands. 

I wanted to do that with the webserver. 
Do you have any other suggestions as to what I can do?

I'm working with a headless device here, so there is no screen. 
I can also use bluetooth I guess, but ultimately that also requires an app. I don't want to have a specific app just to configure the ESP32

xcguang, 2018-08-23:
Hi @Wihan ,
Sorry for that we have no schedule to do it. Maybe you can customize it by yourself in the esp_at_device_ops_struct.  When receiving data, remember the data path, and response data via it when output the result.