espressif/esp-at #123: esp32-wroom restart config BT-Classic.

SamDhoundiyal, 2018-08-21:
Hi,
i am using ESP32-wroom with 520 KB of on-chip SRAM for data and instructions and
448 KB of ROM for booting and core functions. I want to execute BT Classic with set AT Commands

Then I tried to disable some features. and try to config classic. but esp32 restart after this many command and i will receive ready in terminal as receiver when power on module with at-command.

ready
AT+SYSRAM?
+SYSRAM:101168
OK

AT+BTINIT=1
OK

AT+SYSRAM?
+SYSRAM:35076
OK

AT+BTSTARTDISC=0,10,1
OK

ready

Any advise or guidance would be appreciated,

Regards
SAM

xcguang, 2018-09-29:
Maybe it is lack of memory, you can try it using ESP32-wrover

xcguang, 2018-10-11:
Hi,
Please try classic bluetooth on the ESP-wrover32, I will close the issue. If you have any question, please reopen it, or create a new issue, thanks.

closed 2018-10-11 by xcguang