espressif/esp-at #124: AT+BLEGATTCPRIMSRV=0 make "Guru Meditation Error: Core 0 panic'ed (LoadProhibited)"

squallwc12, 2018-08-28:
ESP-WROOM-32
My device is smart bracelet. Model is Hi11pro.
----------
AT+BLEGATTCPRIMSRV=0

ready
----------
in version(Wroom32):1.1.1
I (641) wifi: wifi firmware version: bffcf7f
I (641) wifi: config NVS flash: enabled
I (642) wifi: config nano formating: disabled
I (651) wifi: Init dynamic tx buffer num: 32
I (652) wifi: Init data frame dynamic rx buffer num: 32
I (652) wifi: Init management frame dynamic rx buffer num: 32
I (657) wifi: wifi driver task: 3ffdf6a8, prio:23, stack:4096
I (662) wifi: Init static rx buffer num: 10
I (666) wifi: Init dynamic rx buffer num: 32
I (670) wifi: wifi power manager task: 0x3ffe3d98 prio: 21 stack: 2560
I (705) wifi: mode : softAP (30:ae:a4:15:2f:61)
I (712) wifi: mode : sta (30:ae:a4:15:2f:60) + softAP (30:ae:a4:15:2f:61)
I (716) wifi: mode : softAP (30:ae:a4:15:2f:61)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40133d23  PS      : 0x00060630  A0      : 0x80137ada  A1      : 0x3ffe7540
A2      : 0x00000000  A3      : 0x3ffc76b4  A4      : 0x3ffd6a9c  A5      : 0x3ffd6a9c
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80133d60  A9      : 0x3ffe7520
A10     : 0x3ffd6a70  A11     : 0x40083f38  A12     : 0x3fffff2c  A13     : 0x3ffe7578
A14     : 0x00000000  A15     : 0x3fffff48  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40133d23:0x3ffe7540 0x40137ad7:0x3ffe7560 0x401330b1:0x3ffe75b0 0x4013ad22:0x3ffe75d0

Rebooting...
----------
But, when I used older firmware V1.0.0  worked.

AT+BLECONN=0,"f5:24:fb:8c:9a:1a"

OK
+BLECONN:0,"f5:24:fb:8c:9a:1a"
+BLECONNPARAM:0,128,140,140,0,400
AT+BLEGATTCPRIMSRV=0
+BLEGATTCPRIMSRV:0,1,0x1800,1
+BLEGATTCPRIMSRV:0,2,0x1801,1
+BLEGATTCPRIMSRV:0,3,0x6E400001B5A3F393E0A9E50E24DCCA9E,1
+BLEGATTCPRIMSRV:0,4,0x8E400001F3154F609FB8838830DAEA50,1
+BLEGATTCPRIMSRV:0,5,0xFEE7,1

OK
----------

carlessole, 2019-02-11:
Hi.
I have the same problem using the latest version v.1.1.3.0. 
ESP32 is acting as server and after a succesful connection to this device: [SensorTAG CC650](http://www.ti.com/tool/cc2650stk) if I try to discover all its services ESP32 reboots.
My console log is:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4013be1f  PS      : 0x00060630  A0      : 0x8013fbda  A1      : 0x3ffe7c00  
A2      : 0x00000000  A3      : 0x3ffc7490  A4      : 0x3ffd59a8  A5      : 0x3ffd59a8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8013be5c  A9      : 0x3ffe7be0  
A10     : 0x3ffd597c  A11     : 0x40083e24  A12     : 0x3ffffe0c  A13     : 0x3ffe7c38  
A14     : 0x00000000  A15     : 0x3ffffe28  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4013be1f:0x3ffe7c00 0x4013fbd7:0x3ffe7c20 0x4013b1ad:0x3ffe7c70 0x40142eb6:0x3ffe7c90

Rebooting...
```
The GATT profile of [SensorTAG CC650](http://www.ti.com/tool/cc2650stk) could be found here: 

- http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User%27s_Guide#Gatt_Server

- http://processors.wiki.ti.com/images/a/a8/BLE_SensorTag_GATT_Server.pdf

So, which is the problem? Maybe it's because the GATT profile is not follows the standard defined here: https://www.bluetooth.com/specifications/gatt/generic-attributes-overview?

Thanks

xiewenxiang, 2020-01-09:
Please use the latest release, the problem has been fixed.

closed 2020-03-15 by xcguang