espressif/esp-at #126: Eclipse error *** target pattern contains no '%'.  Stop.

VictorDuma, 2018-09-05:
When I try to compile esp32-at example with the Eclpipse on Win10 I got error /home/mypc/esp/esp-idf/make/component_wrapper.mk:285: *** target pattern contains no '%'.  Stop.
All variables was set according the documentation. Same time I was able to compile with the console. Looks like some conflict with windows style path and unix style. What can I check else?

rochamorelli, 2018-10-03:
I had the same problem when cloning a pre-built repo. Maybe it's not the same problem you have, but I managed to solve it by erasing the `build` folder.

Ralim, 2018-10-04:
Confirming that a `rm -rf` of build worked for me as well.
Issue showed up when moving to WSL from the MYSYS and obviously had old paths cached in the build folder somewhere.

closed 2019-01-16 by FayeY