espressif/esp-at #128: Connect and receive events out of order

ahessling, 2018-09-10:
I am developing a HTTP server with the latest ESP32 AT firmware (036e3556684ba6b721d8df3f96d7358c8147b7a2) using the UART interface connected to a STM32 host controller.

It seems that under some circumstances the TCP CONNECT event is sent after a +IPD event.

Example (sent by the ESP32 module)
...
+IPD,1,369:GET /web/index.html HTTP/1.1 (+ a lot of data)
1,CONNECT
Recv:1000 bytes
SEND OK

The last two lines are put out because of another transmission from the host controller to the ESP32 module in the past, so this should be OK.

In other cases it looks like this, which would also be my expectation:
1,CONNECT
+IPD,1,369:GET /web/index.html HTTP/1.1 (+ a lot of data)

So I wonder why in some cases the CONNECT event is sent out of order. At the moment I use the CONNECT event to internally reset my HTTP parser. But when the events are sent out of order, this behavior would erase the previously received data.

xcguang, 2018-09-29:
Hi，
The ` CONNECT event` is that your client connected the server you created.

ahessling, 2018-10-01:
Hi @xcguang ,
I understand that the CONNECT means that a client connected my server. But the problem is, that under some circumstances, the CONNECT event is sent delayed by the ESP32 module. What I am seeing is that sometimes payload data is sent BEFORE the CONNECT event is put out.

xcguang, 2019-11-06:
Hi,
Is the issue solved? 
Now we add `AT+CIPRECVMODE` and `AT+HTTPCLIENT` command. Maybe you can try them.

closed 2020-03-15 by xcguang