espressif/esp-at #129: AT-Firmware and WiFi/Bluetooth classic protocols

tchouzin, 2018-09-10:
Hello  @xcguang 

I would like to know if the AT-Firmware supports WiFi and Bluetooth classic protocols at the same time? That is, can we run both at the same time? 
I get an error when Wi-Fi is connected and I try to connect the Bluetooth classic.

Regards,

Achille

xcguang, 2018-09-12:
Hi，
If you want to run WiFi and Bluetooth classic protocols, you need a param, because there is no enough ram.

tchouzin, 2018-09-12:
Hello @xcguang ,
What do you mean by  * param?

xcguang, 2018-09-12:
Hi,
"param" is external RAM. Please refer to the document about ESP32-WROVER module at https://www.espressif.com/en/support/download/documents.

xcguang, 2018-09-30:
Hi,
I will close the issue, if you have any question, please reopen it again or create a new issue.

closed 2018-09-30 by xcguang