espressif/esp-at #130: Classic Bluetooth Initialization Error ESP32-WROVER

Arti4ever, 2018-09-19:
Hello, 
I've got an Espressif ESP32-WROVER-B module which is supposed to have external ram to handle wifi/BT/BLE.
I configured the esp32-at firmware to have wifi and BT support with the lastest souces available so far (commit 1a14b55). 

When I try to initialize the classic Bluetooth the module return an error :
AT+BTINIT=1
ERROR

I try disabling the ble at support as said in Issue #110 but this did not work for me.

Can you tell me what I am missing ?

Regards.

Arti4ever, 2018-09-25:
Hi, 
In fact, since the lastest commit 1a14b55, classic bleutooth do not work anymore. 

I've tested with commit 036e3556 and bluetooth initialization worked. 
But AT+BTSTARTDISC return wrong name and sometimes bad characteres. 
and AT+BTSPP is not working...

xcguang, 2018-09-28:
Hi,
Did you configure `Component config > AT > [*]   AT bt command support. `

Arti4ever, 2018-09-28:
Hi,
Yes I did.
Did you test classic Bluetooth with the lastest commit ?

xcguang, 2018-09-29:
Hi,
Please try it again using the latest master branch.
BTW, you need configure `Component config > Bluetooth > Bluetooth controller > (X) Bluetooth Dual Mode`

Arti4ever, 2018-10-01:
Hi, 
Thanks, Initialization works now. 
But this is the only things that works. I will open new issues to be more clear.

closed 2018-10-01 by Arti4ever