espressif/esp-at #131: Empty PKI bin generation prevents further customized_partitions generation

tagunil, 2018-09-21:
Hello,

I'm developing an application which does not require server_cert, server_key etc. It requires just client_ca, so I've removed unused example certs and keys from their raw data directories. After that I've got following build output (**make V=1**):

```
building AT customized partition table at_customize.csv
python /home/tagunil/Projects/Bear/esp32-at/esp-idf/components/partition_table/gen_esp32part.py -q /home/tagunil/Projects/Bear/esp32-at/at_customize.csv /home/tagunil/Projects/Bear/esp32-at/build/at_customize.bin
building AT customized partition table succeed
generating AT customized partition bins to /home/tagunil/Projects/Bear/esp32-at/build/customized_partitions
server_cert: Warning: raw data file not found, generate empty file instead!
AT customized bin generation finished
```

Neither of following PKI bins are generated, and neither are mentioned in the flash command params.
I think this issue is caused by "exit 0" line in generate_pki_bin.sh. That script is sourced, not called, so that line stops generation loop in Makefile.projbuild.

Thanks!

xcguang, 2018-09-29:
Hi,
You can modify `components/customized_partitions/AT_partition_dependency`

tagunil, 2018-09-29:
Thanks, but I have solved my problem by modifying generate_pki_bin.sh script like that:
```
 then
     echo "$PARTITION_NAME: Warning: raw data file not found, generate empty file instead!"
     touch $TARGET_FOLDER/$PARTITION_NAME.bin
-    exit 0
+else
+    echo "generating $PARTITION_NAME.bin: python $TOOL_PATH generate_bin -b $TARGET_FOLDER/$PARTITION_NAME $FILE_CONFIG"
+    python $TOOL_PATH generate_bin -b $TARGET_FOLDER/$PARTITION_NAME.bin $FILE_CONFIG
 fi
-
-echo "generating $PARTITION_NAME.bin: python $TOOL_PATH generate_bin -b $TARGET_FOLDER/$PARTITION_NAME $FILE_CONFIG"
-
-python $TOOL_PATH generate_bin -b $TARGET_FOLDER/$PARTITION_NAME.bin $FILE_CONFIG
```

closed 2018-09-29 by xcguang