espressif/esp-at #132: SSL server verification results are ignored

tagunil, 2018-09-24:
Even after **AT+CIPSSLCCONF=2,0,0** command one can estabilish an SSL connection with a bad server. mbedTLS debug log shows verification errors, but such a connection is successful. This problem is caused by AT core using SSL_VERIFY_PEER flag instead of SSL_VERIFY_FAIL_IF_NO_PEER_CERT one. Unfortunately, there is no way to get verification results by AT commands.

xcguang, 2018-09-29:
Hi,
Could you provide the fail log? We test it, but it seems good.

tagunil, 2018-09-29:
Here it is: [at_ssl_log.txt](https://github.com/espressif/esp32-at/files/2430672/at_ssl_log.txt)

You can see that the verification is failed (my root certificate is from different chain of trust):
> I (119963) openssl: ssl_tls.c:4837 x509_verify_cert() returned -9984 (-0x2700)
> I (119963) openssl: ssl_tls.c:4925 ! Certificate verification flags 8

But then the connection estabilishes as if nothing has happened.

xiewenxiang, 2018-10-25:
Hi @tagunil ，

Can you provide all the steps？

xiewenxiang, 2018-11-12:
@tagunil 
we have fixed it, please pull the latest master branch

ubichsel, 2018-11-14:
After downloading the whole at project with git clone --recursive https://github.com/espressif/esp32-at.git I get an error when compiling it:
...
...
AR build/bt/libbt.a
LD build/esp-at.elf
.../esp32-at/components/at/lib\libat_core.a(at_bleCmd.o):(.literal.at_setupCmdBleAddr+0x4): undefined reference to `esp_ble_gap_clear_rand_addr'
.../esp32-at/components/at/lib\libat_core.a(at_bleCmd.o): In function `at_setupCmdBleAddr':
(.text.at_setupCmdBleAddr+0x1d): undefined reference to `esp_ble_gap_clear_rand_addr'
collect2.exe: error: ld returned 1 exit status
make: *** [.../esp/esp-idf/make/project.mk:411: .../esp32-at/build/esp-at.elf] Fehler 1

xiewenxiang, 2018-11-14:
@ubichsel 

This should be because IDF version is not correct .

ubichsel, 2018-11-14:
I tried with the IDF from AT master branch and with the current master branch of IDF. both won't compile with make (different error messages). Can you please tell me which IDF version is working with the new AT-Lib or even better, upload the working one to this master branch?

eliasfsantos, 2018-12-07:
I am with the same problem. undefined reference to 'esp_ble_gap_clear_rand_addr'

ubichsel, 2018-12-10:
I had to upgrade to the newest build environment (on windows) rom ESP, then i compiled and linked correctly

closed 2019-01-16 by FayeY