espressif/esp-at #136: Classic Bluetooth Scan not working correctly

Arti4ever, 2018-10-01:
Hi,
I'va got an ESP32_Wrover with External RAM enabled.
Classic bluetooth scan (AT+BTSTARTDISC) do not work correctly.

Endeed, sometimes wrong characteres are send to the uart. 
For instance: 
+BTSTARTDISC:98:9c:57:82:a9:2c,øÿÿ?EI P8 lite 2017,-73,2,0x2d0
We see that there are somme characteres "øÿÿ?" who should not be here. 
the real name of the device is "HUAWEI P8 lite 2017"

An other issue is that the name buffer is not clear between two messages.
For exemple, I get this:
+BTSTARTDISC:00:06:66:77:7e:ca,øÿÿ?EI P8 lite 2017,-48<CR><LF>
+BTSTARTDISC:00:06:66:77:7e:ca,PRECIA_i5-7ECA 2017,-43<CR><LF>
=> the "2017" should not be here for the "PRECIA_i5-7ECA" device. It seems it comes from the previous device "HUAWEI P8 lite 2017"

And an other issue is that MAC address do not always match the device name... 
In the previous exemple, both devices have the same MAC address but in fact HUAWEI have "98:9c:57:82:a9:2c" and "PRECIA_i5-7ECA" have "00:06:66:77:7e:ca"

So there is many many bugs in bluetooth scanning.

Arti4ever, 2018-10-16:
Did you find something ?

Arti4ever, 2018-11-11:
up

xiewenxiang, 2018-11-12:
@Arti4ever 
Please pull the latest master branch

closed 2018-11-13 by Arti4ever