espressif/esp-at #137: Boot-loop when HSPI is used as AT interface

TinghuiWang, 2018-10-07:
I have successfully compiled and verified the `esp32-at` binaries for my ESP-WROOM-32 module with AT interface set to Uart0 and Uart1. However, when I try to test the AT commands using HSPI interface, fatal error occurs that brings ESP32 module into a boot loop.

From the Uart0 log, I am concerned about the following error messages:

```
E (653) HSPI-AT: Cannot get read data address.
...
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400f1e40
file: "/builds/rd/esp-at-core/components/at_core/at_port.c" line 711
func: initialise_wifi
expression: esp_event_loop_init(at_wifi_event_handler_cb, NULL)

Backtrace: 0x40093ae0:0x3ffba9a0 0x40093eb0:0x3ffba9c0 0x400f1e40:0x3ffba9e0 0x400f28e3:0x3ffbaaa0 0x400d4161:0x3ffbaae0 0x400d2408:0x3ffbab00 0x40090f29:0x3ffbab20

Rebooting...
```

I am wondering if anyone has experienced this error before or if anyone can advice on what I need to look into. Thanks indeed.

Here is the full log with output level set to `verbose`.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6256
ho 0 tail 12 room 4
load:0x40078000,len:9404
load:0x40080400,len:7128
entry 0x400807a0
I (30) boot: ESP-IDF v3.2-dev-1175-g11b444b8f 2nd stage bootloader
I (31) boot: compile time 16:58:37
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (75) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (82) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (90) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (97) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (105) boot: End of partition table
I (109) boot: No factory image, trying OTA 0
I (114) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2b7bc (178108) map
I (185) esp_image: segment 1: paddr=0x0012b7e4 vaddr=0x3ffbdb60 size=0x03508 ( 13576) load
I (191) esp_image: segment 2: paddr=0x0012ecf4 vaddr=0x3ffc1068 size=0x00000 (     0) load
I (192) esp_image: segment 3: paddr=0x0012ecfc vaddr=0x40080000 size=0x00400 (  1024) load
I (202) esp_image: segment 4: paddr=0x0012f104 vaddr=0x40080400 size=0x00f0c (  3852) load
I (212) esp_image: segment 5: paddr=0x00130018 vaddr=0x400d0018 size=0xf54b4 (1004724) map
I (571) esp_image: segment 6: paddr=0x002254d4 vaddr=0x4008130c size=0x15814 ( 88084) load
I (608) esp_image: segment 7: paddr=0x0023acf0 vaddr=0x400c0000 size=0x00064 (   100) load
I (608) esp_image: segment 8: paddr=0x0023ad5c vaddr=0x50000000 size=0x00000 (     0) load
I (627) boot: Loaded app from partition at offset 0x100000
I (628) boot: Disabling RNG early entropy source...
I (628) cpu_start: Pro cpu up.
I (632) cpu_start: Single core mode
V (636) memory_layout: reserved range is 0x3f42b794 - 0x3f42b7dc
D (642) memory_layout: Checking 11 reserved memory ranges:
D (648) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (654) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (661) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (667) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (673) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (680) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffcc3f0
D (686) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
[... 286 lines cut ...]
D (951) nvs: nvs_open_from_partition NVM_009 0
D (955) nvs: nvs_get auth3 1
D (958) nvs: nvs_open_from_partition NVM_009 0
D (963) nvs: nvs_get auth4 1
D (966) nvs: nvs_open_from_partition NVM_005 1
D (970) nvs: nvs_get enable 1
D (973) nvs: nvs_get type 1
D (976) nvs: nvs_get_str_or_blob r_ip
D (980) nvs: nvs_get r_port 4
D (983) nvs: nvs_get l_port 4
D (986) nvs: nvs_get change_mode 1
D (989) nvs: nvs_get keep_alive 2
D (993) nvs: nvs_open_from_partition NVM_013 0
D (997) nvs: nvs_get apmode 4
D (1001) nvs: nvs_close 11
D (1003) nvs: nvs_open_from_partition NVM_013 1
D (1008) nvs: nvs_set apmode 4 0
D (1012) nvs: nvs_close 12
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400f1e40
file: "/builds/rd/esp-at-core/components/at_core/at_port.c" line 711
func: initialise_wifi
expression: esp_event_loop_init(at_wifi_event_handler_cb, NULL)

Backtrace: 0x40093ae0:0x3ffba9a0 0x40093eb0:0x3ffba9c0 0x400f1e40:0x3ffba9e0 0x400f28e3:0x3ffbaaa0 0x400d4161:0x3ffbaae0 0x400d2408:0x3ffbab00 0x40090f29:0x3ffbab20
```

xcguang, 2018-10-08:
Hi,
Please remove `esp_at_module_init(CONFIG_LWIP_MAX_SOCKETS - 1, version);` in `main/interface/hspi/at_hspi_task.c`, and we will fix it ASAP.

TinghuiWang, 2018-10-08:
Thanks for your response.
Removing `esp_at_module_init` does remove the fatal error of `initialise_wifi`, however, the SPI is still not functioning properly.

I have the following two questions:

1. SPI pin configuration:
When HSPI is in slave mode, `MOSI` is mapped to `GPIO12` (`HSPIQ`), `MISO` is mapped to `GPIO13` (`HSPID`).
However, the `SCLK` and `CS` signals are not mapped to the same pins as described in ESP32_TRM (v3.8) document.
`SCLK` is mapped to `GPIO15` in the project, which is labeled as `HSPICS0` in ESP32_TRM and ESP-WROOM-32 datasheet.
`CS` is mapped to `GPIO14` in the project, which is labeled as `HSPICLK` in ESP32_TRM and ESP-WROOM-32 datasheet.

Which one is correct?

2. HSPI-AT error message still occurs.

During the log, there is still the following error message:

```
E (653) HSPI-AT: Cannot get read data address.
```

yake827, 2018-10-09:
Hi @TinghuiWang 1. SPI pins use iomux map [SPI iomux](https://github.com/espressif/esp-idf/blob/master/components/driver/spi_common.c#L195), so you can use any free IO implementation. 2. This Log is normal, we'll get rid of it in the next patch.

TinghuiWang, 2018-10-10:
@yake827 Thank you very much for clearing it up.

closed 2018-10-11 by xcguang