espressif/esp-at #138: Classic Bluetooth SPP Connection issue

Arti4ever, 2018-10-08:
Hi,
I'va got an ESP32_Wrover with External RAM enabled.
SPP connection (AT+BTSPPCONN) seems not working.

I enabled bluetooth :
AT+BTINIT=1
OK
then I enabled SPP :
AT+BTSPPINIT=1
OK

but when I trying to connect to my device I get an Error: 
AT+BTSPPCONN=0,0,0,"00:06:66:77:7e:ca"
ERROR

I tryied with different <sec_mode> (as describe in doc) but I can't get it work. It always return "ERROR"

By the way, How do we specify the PIN code for connection ?

Arti4ever, 2018-11-11:
up

xiewenxiang, 2018-11-12:
@Arti4ever 

Please pull the latest master branch. Before compile the project, you need enable the BT command and  set  bluetooth controller dual mode.

`make menuconfig -> component config -> Bluettoth -> Bluetooth controller -> Bluetooth controller mode -> Bluetooth Dual Mode`

vagr02, 2018-11-21:
Hi, i am facing same problems with SPP. I am using development kit LyraT with WROVER module. I set all configurations in menuconfig - enabled PSRAM ( i can see more RAM by command AT+SYSRAM?  so, it works)  . enabled Bluetooth and AT commands for Classic Bluetooth, but still i can see ERRORs only when i use AT+BTSPPCONN , AT+BTSPPSTART. AT+BYSPPSEND. Only working command is AT+BTSPPINIT, it shows OK. Could you please tell me correct initialization of SPP connection? 
at moment i do:
 AT+BTINIT=1
 OK
AT+BTNAME="ESP32_BT_AT"
 OK
AT+BTSPPINIT=1
OK
AT+BTSPPCONN=0,0,0,"xx:xx:xx:xx:xx:xx"   // ( i used valid remote address here)
ERROR
AT+BTSPPSTART
ERROR

vagr02, 2018-11-21:
Great, it works! Thank you very much. I see, there is AT+BTSPPINIT=2 (i didnt find this value in documentation ESP32_AT_Commands_Set.md) I have one more question to this topic. How can i change security mode in slave mode? I need to use legacy pin code mode 4-digit pin. Can i use for this AT+BTSPPCONN command somehow?

xiewenxiang, 2018-11-22:
Hi @vagr02 
We are working on these commands,  It should be done by next week.

vagr02, 2018-11-30:
Hello, i hope it will be possible to set different security modes. I am working for producer of digital multimeters (Gossen Metrahit series) and we plan use ESP32 modules as communication interface for Wifi and Bluetooth classic. We would like to replace current Bluetooh module from Microchip RN4678. Therefore we need to implement in ESP32 some of features of RN4678. One of them is this 4 pin security mode and other is possibility to change UUID from default Serial Port 0x1101 to a different number. RN4678  uses a command for this , but in beginnig even setting in menuconfig, or possibility to change this manually in a sourcecode would be enough for us.

xiewenxiang, 2018-11-30:
@vagr02 

We will release this feature today.

vagr02, 2018-12-05:
I was testing new commands and i am not sure how to use them to be able to enter legacy PIN code. I think i should use command:  AT+BTSECPARAM=3,1,"1234"  - fixed PIN, but when PC connects to ESP32 it shows variable (randomly generated) PIN. This happens when i am using USB-Bluetooth adaptor v4.0 EDR. When i use older v3.0 Bluetooth adaptor i can see dialog for entering PIN code. This happens in Win 10.

xiewenxiang, 2018-12-06:
@vagr02 

you need to disable the simple pair

[disable_ssp.patch.txt](https://github.com/espressif/esp32-at/files/2651344/disable_ssp.patch.txt)

xiewenxiang, 2018-12-06:
@vagr02 
If you want to change the SPP UUID, you can change this manually in the source code.

the path is `esp-idf/components/bt/bluedroid/api/esp_spp_api.c `

```c
static const uint8_t  UUID_SPP[16] = {0x00, 0x00, 0x11, 0x01, 0x00, 0x00, 0x10, 0x00,
                                      0x80, 0x00, 0x00, 0x80, 0x5F, 0x9B, 0x34, 0xFB
                                     };
```

vagr02, 2018-12-10:
I tried to change UUID in esp_spp_api.c , but it still shows identification 0x1101. I cleaned flash, build completely new firmware but UUID is still same. I am not sure if there in sourcecode is not UUID rewritten somewhere.

xiewenxiang, 2019-01-23:
Does BT SPP working now?

closed 2019-01-23 by FayeY