espressif/esp-at #139: AT+CIPSTART response time

mrbiondo, 2018-10-10:
Hello everybody, 
when the AT+CIPSTART command fails to connect to a remote server, it returns an ERROR message after a long time -about 10 seconds in my case. Is there a way to modify the response time of that command?

Also, when the AT+CWJAP command fails to connect to an access point, the response time il long -about 20 seconds, in my case. Is that a problem of access point connection? Or, the same question as above.

Thanks for reply,

xcguang, 2018-10-11:
Hi,
`AT+CIPSTART` and `AT+CWJAP` are the block command, they need take some time to finish it.
`AT+CIPSTART` will retry some times in the TCP/IP protocol.
`AT+CWJAP` will retry if joining fail, until 15 seconds timeout.

mrbiondo, 2018-10-11:
Hi, 
i saw they are block command. You say, there are no chances to change their responce time for both of them, right?

closed 2019-01-16 by FayeY