espressif/esp-at #142: "Guru Meditation Error"

mrbiondo, 2018-10-23:
Hi, 
that seems to be a known problem, but I do not undestand whether it is fixed.

ASSERT_PARAM(131072 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

My scenario is:
-WiFi init in station mode
-run the command AT+CWLAP="myssid" every  10 seconds, and "myssid"  is never available 
-BLE init in peripheral role and connected to a central
-run the command AT+ADVSTART every minutes

xiewenxiang, 2018-11-12:
@mrbiondo 
Please provide the version information you used， you can get it by this command `AT+GMR`

mrbiondo, 2018-11-12:
Here it is.
AT version:1.2.0.0-dev(6e68431 - Sep 21 2018 09:16:33)
SDK version:v3.1-34-gcea4d3132

xiewenxiang, 2019-01-22:
@mrbiondo 
This issue will be fixed in the next update.

mrbiondo, 2019-01-23:
@xiewenxiang 
Is that fixed with the latest master branch? Thanks.

xiewenxiang, 2019-01-23:
This bug has been fixed, but the AT has not been updated. If you need, we can upgrade the release /v1.2 branch, but it hasn't been fully tested.

closed 2020-01-09 by xcguang