espressif/esp-at #143: is there way to store/upload SSL Client Certificate via AT Command on ESP32 MCU

valoni, 2018-10-26:
is there way to store/upload SSL Client Certificate on ESP32 MCU 
 via AT Commands with no need for Downloader Tools

xcguang, 2018-10-29:
Hi,
Maybe `AT+SYSFLASH` is helpful for you.

valoni, 2018-10-29:
please can you provide a little sample 

how to store certificate using this AT+SYSFLASH

0.delete existing cert,key,...
1.upload and save new cert,key,....

xcguang, 2018-10-29:
```
AT+SYSFLASH=<operation>,<partition>,<offset>,<length>
Function:
Read/write the user partitions in flash.

- <operation>:
0: erase sector
1: write data into the user partition
2: read data from the user partition
- <partition>: name of user partition
- <offset>: offset of user partition
- <length>: data length
```
For example：
// read 4096 bytes from the "ca_data" partition offset 0.
AT+SYSFLASH=2,"ca_data", 0,4096
// write 4096 bytes to the "ca_data" partition offset 0.
AT+SYSFLASH=1,"ca_data",0,4096
// erase 1 sector from the "ca_data" partition offset 0.
AT+SYSFLASH=0,"ca_data",0,4096

And about SSL Client Certificate format for ESP32 AT, please refer to https://github.com/espressif/esp32-at/blob/master/tools/README.md

closed 2019-01-16 by FayeY

espressifelse, 2020-05-21:
> // write 4096 bytes to the "ca_data" partition offset 0.
> AT+SYSFLASH=1,"ca_data",0,4096

In this example what are the data that are being written to partition ca_data? I don't see a <data> parameter. Where do I enter the data that I want to write? For example, suppose I want to write a binary to an app partition. I would want to write the contents of hello-world.bin to partition ota_0. How can I do this using AT+SYSFLASH

valoni, 2020-05-21:
> In this example what are the data that are being written to partition ca_data? I don't see a data parameter. Where do I enter the data that I want to write? For example, suppose I want to write a [...]

https://www.espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf

page 97

[image]

espressifelse, 2020-05-21:
Thanks, yes I see now how it works. After sending the command the UART waits for the data to follow. 
>