espressif/esp-at #145: BLESPP Error

jamesfowkes, 2018-11-14:
I am trying to use BLE SPP. I have used the flash download tool to put:

`est-at.bin` at `0x100000`
`ATBleService.bin` at `0x21000`
`at_customize.bin` at `0x20000`

`est-at.bin` and `at_customize.bin` are from the v.1.1.2 release
`ATBleService.bin` generated from GattServiceExample.csv in `esp32-at/tools` using `BLEService.py`.

When I try to enable BLE SPP with AT commands as per the example in the `ESP32_AT_Instruction_Set_and_Examples__EN.pdf` file:

```
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xED2B4E3B2820492F9507DF165285E831,0x08
+BLEGATTSCHAR:"char",1,2,0xED2B4E3C2820492F9507DF165285E831,0x10
+BLEGATTSCHAR:"desc",1,2,1,0x2902

OK
AT+BLEADVSTART

OK
+BLECONN:0,"57:0b:e8:d7:df:6a"
+BLECONNPARAM:0,0,0,6,0,2000
+BLECONNPARAM:0,0,0,39,0,2000
+BLECONNPARAM:0,0,0,6,0,2000
+BLECONNPARAM:0,0,0,39,0,2000
AT+BLESPPCFG=1,1,7,1,5

OK
AT+BLESPP

ERROR
```

I do not know what the issue is. Do I need a different set of GATT parameters?
How can I get BLE SPP working?

xiewenxiang, 2018-12-06:
@jamesfowkes 

1. The client need to write the ccc
2. Parameter error
    Try this `AT+BLESPPCFG=1,1,2,1,1`

closed 2019-01-16 by FayeY

saravanangmail, 2019-01-31:
Yeah i had tried AT+BLESPPCFG=1,1,2,1,1 its not working

xiewenxiang, 2019-01-31:
@saravanangmail 
Could you provide your `ble_data.bin`

saravanangmail, 2019-01-31:
i had used gattservice bin instead ble_data.bin
[image]
please change the extension to bin, since we couldn't able to upload bin file

KingChen84, 2020-01-10:
@saravanangmail before you AT+BLESPPCFG=1,1,7,1,5. you must connect server "ESP32", and than, open 0xC306 char, and Listen for "notifications". And than, send "AT+BLESPP", it will work.