espressif/esp-at #147: compiled with FS but i dont know if it works

vagr02, 2018-11-22:
Hi,
I did some tests of AT+FS command with LyraT board and 4GB Micro SD card inserted and formated to FAT filesystem. I compiled software (last master branch)  with enabled AT-FS (i didnt find any other settings for SD card in menuconfig). Unfortunately i cannot read and list files on SD card. I dont know what i am doing wrong?  I have SD card with one file: esp32tst.txt and i get those results:

AT+FS=0,4,"."
+FS:test.txt     <--- there is not such a file on SD card
OK

AT+FS=0,2,"esp32tst.txt",0,100
open /fatfs/esp32tst.txt fail
ERROR

vagr02, 2018-11-30:
Maybe just wrong parameters of command or compilation settings?  :)

xcguang, 2018-11-30:
Hi，
Now AT+FS cannot support SD card.

vagr02, 2018-11-30:
Thank you for information. I hope SD card will be supported in near future. It is useful function :-)

vagr02, 2019-01-24:
Hello, is there something new about filesystem and SD card support?  :-)

toko1, 2019-02-28:
Hi,
is SD card already supported in AT+FS? I need access to the SD card via AT commands .

jpazderka, 2019-03-28:
Hi,
it would be very nice to have this AT support - is there anything new about it?