espressif/esp-at #149: Custom UUID for Classic Bluetooth

vagr02, 2018-12-05:
Hello.
We would like to replace current Bluetooh module from Microchip RN4678. Therefore we need to implement in ESP32 some of features of RN4678. One of them is possibility to change UUID from default Serial Port 0x1101 to a different number. RN4678 uses a command for this , but in beginnig even setting in menuconfig, or possibility to change this manually in a sourcecode would be enough for us. Is there some possibility to change it easily?

xiewenxiang, 2018-12-06:
We do not support such a configuration at present and may consider it in the future

xiewenxiang, 2018-12-06:
```c
static const uint8_t  UUID_SPP[16] = {0x00, 0x00, 0x11, 0x01, 0x00, 0x00, 0x10, 0x00,
                                      0x80, 0x00, 0x00, 0x80, 0x5F, 0x9B, 0x34, 0xFB
                                     };
```

xiewenxiang, 2018-12-06:
the path is `esp-idf/components/bt/bluedroid/api/esp_spp_api.c`

vagr02, 2018-12-06:
Thank you, very much. This solution - change in source code should be good enough for us. I understand it is not standart UUID, but unfortunately it is a wish of our other software department.

vagr02, 2018-12-07:
I tried to change UUID in esp_spp_api.c , but it still shows identification 0x1101. I cleaned flash, build completely new firmware but UUID is still same. I am not sure if there in sourcecode is not UUID rewritten somewhere. I am trying to find it.

FayeY, 2019-01-23:
Sorry that the UUID is a default one defined in the BLE spec, cannot be changed. You may need to implement it by yourself. Sorry for the inconvenience. Thanks.

closed 2019-01-23 by FayeY

jpazderka, 2019-03-28:
Hi - We have the same problem - I tried to change the code according your comment from dec. 6. 2018. Unfortunately it doesnt work. May be, there are used another constants in the source code or in libraries.  Ca you help us to find it?  Thanks