espressif/esp-at #155: Classic Bluetooth SPP slave with PIN

TomaszKosecki, 2019-01-10:
Hi,

I have problem with setting classic BT SPP slave to be secured with PIN.
Is it possible to achieve this with AT commands?
I was able to set ESP32 to be classic BT master and connect to slave which was secured with PIN,
i did it with command: "AT+BTSECPARAM=3,1,"some_pin".
But it seems it works only with master. Right?
I can't do this for slave.

I saw issue #138 and I used patch provided there and then:
- I had to input PIN to be able to connect to ESP32 from my mobile :)
but:
- when I try to connect to ESP32 from another ESP32 it is still possible without PIN :(
- when I switch the role to master, scanning doesn't work (one of our devices has to change roles in runtime)  :(

Is this really impossilbe to set ESP32 as BT classic SPP slave with PIN using AT commands or I just make some mistake?
If this is the case are You plannning to introduce such command/commands?

erdemp, 2019-01-16:
Hi,
I try to define ESP-WROOM-32 as master to connect to a BT classic device. But I cant find the command you said "AT+BTSECPARAM" in AT commands list. How do you use this command. Should I have to enable something?

xiewenxiang, 2019-01-16:
By default, classic bluetooth is not enabled. Do you have it enabled in menuconfig？
Please provide the version you are using now
`AT+GMR`

erdemp, 2019-01-16:
I did not enable it. I use ESP-WROOM-32 module with default AT firmware and here are my details. 

AT+GMR
AT version:1.1.3.0(5a40576 - Nov 28 2018 12:50:55)
SDK version:v3.0.7
compile time:Dec 21 2018 09:04:56
Bin version:1.1.3(WROOM-32)

AT+SYSFLASH?
+SYSFLASH:"ble_data",64,1,0x21000,0x3000
+SYSFLASH:"server_cert",64,2,0x24000,0x2000
+SYSFLASH:"server_key",64,3,0x26000,0x2000
+SYSFLASH:"server_ca",64,4,0x28000,0x2000
+SYSFLASH:"client_cert",64,5,0x2a000,0x2000
+SYSFLASH:"client_key",64,6,0x2c000,0x2000
+SYSFLASH:"client_ca",64,7,0x2e000,0x2000
+SYSFLASH:"factory_param",64,8,0x30000,0x1000
+SYSFLASH:"fatfs",1,129,0x70000,0x90000

OK

erdemp, 2019-01-16:
How can I use menuconfig? Is there any document for that?

xiewenxiang, 2019-01-16:
You need to use the latest master branch, and enable BT Command in menuconfig.

```
Component config -> AT -> [*] AT bt command support.
```

xiewenxiang, 2019-01-16:
[ESP32_AT_Classic_Bluetooth.md](https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Classic_Bluetooth.md)

xcguang, 2019-11-12:
Close because of no update for a long time, if you have any question, please reopen it again.

closed 2019-11-12 by xcguang